maelstrom2001/ArcticFox #206: [ENHANCEMENT] Battery Profiles

Dil3mm4, 2017-03-10:
Hello again. 
It would be nice to have a more customizable menu in battery profiles. 
At the moment we can only modify the "Custom" profile and save it. 
Would be nice to have some options to add more than one custom profile and to edit the name too. 
The delete and modify features for already bundled profiles would be nice too, so the user can decide which profile to mod or delete without wasting space in the custom firmware.

TBXin, 2017-03-10:
Hello,

We are think at this moment about to remove all battery presets from firmware and get to users only custom profile which will contain by default GEN points.

Current presets are almost never satisfy the users what causes many of the same issues.
Unfortunately people often do not understand how these presets are works, and that they need to adjust preset for specific battery manually.

1881hc, 2017-03-11:
What does battery preset do besides define the voltage for each % so the battery level indicator may be more accurate?  Does choosing a particular battery change the performance of  the mod?

GnFlala, 2017-03-11:
it does nothing more than the visual . 
I think only the cutoff voltage that you see in custom profile does something real (at least it should work in all profiles and not just custom, maybe it does) but a dev will tell us if that stands true.

TBXin, 2017-03-12:
@GnFlala you are right.

Each profile consists of 11 points on and cut-off.
Points are required to calculate % (just visual part)
cut-off is required to calculate weak-battery, battery low warnings.

Profile does not affect charging or power consumption, just visual.

GnFlala, 2017-03-17:
Since cut-off is global and does something then maybe it should be outside battery profiles. But that is nfe toolbox and not this firmware. 
That would make more sense.

sala91, 2017-04-09:
It would be nice to get HΩ LIFE batterys and MXJO as presets.

maelstrom2001, 2017-04-12:
> add more than one custom profile

Agreed. In upcoming release we'll left only one fixed preset, GEN (equivalent to stock discharge profiles for changeable and in-built cells). And we'll add 3 custom battery profiles with editable names, like TFR and curves sets.

pikachukaki, 2017-04-13:
It could be nice if someone with more knowledge could write a tutorial on how we can manually create a profile specific for our batteries.

Dil3mm4, 2017-04-13:
@sala91 it's pretty much impossible to cover up every single 18650/26650 battery with a custom profile per battery and still, it's wasted space for the ROM itself.
@pikachukaki I can explain how I do mines, but I use an external charger which tells me the percentage. (no guarantees on result tho) 
1. Set CUSTOM profile in NFE
2. Use a 100% charged battery
3. On each 10% drop (to be sure that it's a -10% shut off the box and then turn it on again) insert your battery in the external charger and write up the percentage that the ext charger states. 
4. At the end you will have to modify the CUSTOM profile in NFE with the values obtained from the ext. charger

pikachukaki, 2017-04-13:
I have a charger with leds so I can't measure them properly

TBXin, 2017-04-13:
@pikachukaki no wonder this option is for advanced users ^^ i mean that you must have various devices to measure discharge of the battery such as multimeter or charged with voltmeter or DNA200+ board.
GEN profile is more than enough for 99% of users.

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001