maelstrom2001/ArcticFox #208: Weak battery resets mod at high wattage (Evic VTC Mini)

phildabill, 2017-03-11:
Hello,

I sometimes use a VTC mini or DNA device to test fake HG2s.  Set the device to 75W with a 0.2ohm coil and fire away. If it can handle 5x 6-7seconds pulls the battery is most likely not a fake.

After flashing to ArticFox, The device resets and gives an "atomizer low" message when running this test with a fake HG2. Using the same atty/battery on an unflashed VTC Mini gives the "weak battery" error.

I assume there is a bug somewhere causing this? Or is this by design?

Thanks

traviswparker, 2017-03-13:
That's not a reliable test of anything. 

My VTC Mini will always reset if the battery voltage sags below 3.3v, and that is with a genuine HG2 or 25R. It's a hardware limitation. The chip needs 3.3v to run. 

AFox might be applying higher initial power and pulling the battery down too hard.

phildabill, 2017-03-14:
kk..makes sense. Why is AF pulling more power initially then?

TBXin, 2017-03-14:
Firmware will work until battery voltage is higher than ~2.8 volts under load.

closed 2017-03-29 by maelstrom2001