maelstrom2001/ArcticFox #209: clock bug

mirko93s, 2017-03-11:
I have an evic vtwo mini, with rtc chip...
with 170201 firmware... I set clock often, but every next day the clock is wrong

TBXin, 2017-03-11:
Turn on X32.

mirko93s, 2017-03-12:
Thanks for answer... I activated x32 yesterday... after 15 hours the clock is about 17 minutes in adavance...

Inviato da iPhone

TBXin, 2017-03-12:
Bad for you. Most likely your device has a defective RTC-module and we can't fix this programmatically.

mirko93s, 2017-03-12:
With original fw and with older arcticfox fw it works good...

Inviato da iPhone

TBXin, 2017-03-12:
We have not changed anything in the RTC.
If it worked before it should now.

You can try to reset your device settings, restart device and configure it again from scratch.

mirko93s, 2017-03-12:
Maybe I have to adjust clock speed... ehat value i should set?

Inviato da iPhone

TBXin, 2017-03-12:
May be you have LSL on earlier?

mirko93s, 2017-03-12:
I disabled lsl beacuse vtwo mini has rtc crystal... if i am not mistaking... how can i set rtc speed properly?

Inviato da iPhone

TBXin, 2017-03-12:
RTC Speed does not affect anything when X32 is ON or LSL is ON.

X32 = On - time is calculated by integrated RTC Module.
LSL = On - time is calculated by primary processor.

X32 = Off, LSL = Off - time is calculated in standby mode by LIRC module, for which is RTC Speed is designed.

mirko93s, 2017-03-12:
Ok thanks for explanation... I try to reset my device... Thank you

Inviato da iPhone

maelstrom2001, 2017-03-12:
@mirko93s Enter Main Menu -> Settings -> RTC -> RTC Speed. Normally, if RTC using hardware 32768 crystal there should be label "X32 On" instead of value. If you see there some value such as 33800, it indicates that firmware was failed to initialize hardware module, something goes wrong and oscillator was switched to LIRC. In this case RTC will not be accurate, and the solution is to use software correction with forcing X32 to Off and LSL to On. Or try to reset box, flash stock firmware and go back to AF, reinstall battery and play with settings while you get "X32 On" in RTC Speed dialog - that way you'll get full hardware support.

mirko93s, 2017-03-12:
Thanks a lot! I'll try this solution and I'll tell you if i solve problem... have a nice day

Inviato da iPhone

maelstrom2001, 2017-03-12:
@mirko93s And a little addition to previous post :) 
Try to set LSL = Off, X32 = On in Expert Menu, select Exit item, wait for 10 seconds to be sure that settings was written to dataflash, and remove battery. Wait about 2 minutes, install battery and go to Expert Settings.
If you find that X32 option goes to Off that is the sign that hardware was not initialized properly.
If this option will remain On after power-on - good news, hardware RTC is working properly now.

mirko93s, 2017-03-12:
Ok I set x32 on, lsl off, removed battery, wait 2 minutes, insert battery... now in rtc speed it says x32 on... I think it works good now... Thanks for help man!

Inviato da iPhone

maelstrom2001, 2017-03-12:
> now in rtc speed it says x32 on

Yep, now the clock should be accurate as in stock firmware. Cheers! :)

closed 2017-03-19 by TBXin