maelstrom2001/ArcticFox #210: Dry Cotton

Prole97, 2017-03-12:
You can't improve on the firmware the advice "Dry cotton" when the ohm of the coil go up (kanthal/tc)

ZenFree, 2017-03-12:
Are you asking for temperature control for Kanthal coils?

Prole97, 2017-03-12:
Nope, i ask for an advice when the cotton is dry, if the cotton is dry the ohm of the coil grown up

Inviato dal mio telefono Windows 10

ZenFree, 2017-03-12:
The coil resistance increase is directly proportional to the coil temperature. When coil is saturated then some amount of power is used to evaporate the liquid which slows down the rise of coil temperature. On the dry coil all power is used to heat the coil so the coil is warming up faster. TC mode controls how much resistance increase to allow. It approximates temperature based on TCR curve. The Kanthal coil changes resistance too little for TC to calculate temperature.
Does it answer your question?

Prole97, 2017-03-12:
Yes, thank you

Inviato dal mio telefono Windows 10

StillVoidingWarranties, 2017-03-13:
Mods cannot measure humidity of the wick or temperature of the coil directly, they do not have sensors for that. They try to calculate the coil's temperature from its resistance - if it is made of a material which has a negative temperature coefficient (which kanthal does not).

Sanaki, 2017-03-13:
> if it is made of a material which has a negative temperature coefficient (which kanthal does not).

For the sake of accuracy (though it changes nothing for these purposes), kanthal does indeed have a measurable TCR. It's just so obscenely small that as of yet -no one- has managed to make an affordable and compact way to actually read it.

closed 2017-03-14 by TBXin