maelstrom2001/ArcticFox #211: Request: "Puff limit" resettable by 3 fast click

prematurata, 2017-03-12:
Hello, 

I think it would be really important for many people, an option that allows user to limit tendency to compulsive vaping, that e-cigs facilitates.
a good way is by modifying or extending the puff limit parameter that already exists. Allowing for example to set a "puff number limit", which if exceeded device stops and continue to show the message "puff exceeded, 3 fast click to reset". So reset will restore previous max puff number limit, and the ability to puff, In this way user can have an increased perception of the quantitative level of vaping

Then I know that devices sometimes can't measure time correctly, so in the future you could then evaluate the additional option of Time limit parameter in conjunction with the puff limit parameter, such as stricter mode that if flagged, when exceeded puffs limit, does not allow the Reset for a previously set number of minutes
I really hope you to consider this request, very important to me and thank you in advance for your attention and your work!!

Dil3mm4, 2017-03-12:
#176

prematurata, 2017-03-12:
i did'nt see, ya a lot of people need this, really hope @maelstrom2001 can help us

wulfgrim, 2017-03-13:
Wouldn't it be realatiitly easy to 1) up your nicotine 2) decide to stop vaping after 5 minutes of chain vaping? 3) set the puff counter to show on the screen so you can see how many puffs you've taken?

prematurata, 2017-03-13:
many vapers normally already do it and more, here we're talking about other important support options

andreivoinea93, 2017-03-13:
...can we stop overcomplicating everything already please?

prematurata, 2017-03-13:
in effect this's a Request to @maelstrom2001 , so anyone can tell an opinion if useful and relevant but at the end, only the coder can decide about it.

Dil3mm4, 2017-03-13:
@wulfgrim 
1) That's not a problem of nicotine. 
2) The problem it's that arbitrary stop doesn't come that easy, expecially for newbies, because they're not (neither am I) watching the clock every sec.
3) Again, that's not a problem of puff counting, but it's the same old story about the "endless cigarette".
Mods (and vaping in general) are some useful instruments to (maybe and possibly) quit smoking. 
And there are some people that are in need of something that ends the vape session as it happens with normal cigs when they burn out naturally.
@andreivoinea93 Can you stop saying bullshits? Someone has shared an idea here. And there are 2 issues on this matter. So if you don't care about something, just don't comment the issue opened.
If you don't want to have things "overcomplicated" as you state, you shouldn't even use a custom firmware and stay on OFW.
Be respectful.

prematurata, 2017-03-13:
@Dil3mm4 i agree, this is exactly the core concept. Main post upgraded, i'll make a donation if this work will be realized.

prematurata, 2017-03-14:
Thanks very much @TBXin , we stay tuned

CCCheshire, 2017-03-21:
'Set puffs limit' can be helpful. It can be simlier - after riching puff limit device can simly blocks (like normal block (sleep) on multiclick) and resets puffs counter.

prematurata, 2017-03-21:
Sounds good, interesting

cmock, 2017-03-23:
One more "thumbs up" for this request.

I talked to smokers that don't want to vape because smoking is a time management help for them. Like "I'll have a smoke break now" means a 5 minute (or however long a cig takes) break.

@maelstrom2001 : would you accept a pull request implementing this feature?

maelstrom2001, 2017-03-23:
Well, this issue is labeled "Questionable Feature", not "Won't Do" :) Everything has its time. We are open to all opinions and suggestions and do not ignore anything. In the plans is a big update in the upcoming release, so please be patient and stay tuned :)

andreivoinea93, 2017-03-24:
Little offtopic but I can't wait for the next big update, can we get a sneak preview of what it's about? :D