maelstrom2001/ArcticFox #214: Eleaf pico dual

Tomozaki123, 2017-03-13:
Hi.
Great frimware, but in pico dual in wattage mode you can't set w over 100. I have pico dual and rt 25 or melo 300.

maelstrom2001, 2017-03-13:
What resistance does the atomizer have?

StillVoidingWarranties, 2017-03-13:
Cannot confirm (170222), my Pico Dual lets me set 168 W on a 0.48 Ohm coil.

Tomozaki123, 2017-03-14:
I have Eleaf ERLQ 0.15Ohm coil and Eleaf RT 25 ( or Eleaf Melo 300) atomizer.

TBXin, 2017-03-14:
@Tomozaki123 you can't set more that ~100W for 0.15 Ohms coil because of 25A limitations on the PCB.
To get more power you need to increase your coils resistance.

Here screenshot of the Ohms law calculator:
[image]

maelstrom2001, 2017-03-14:
@Tomozaki123 update firmware to [ the latest nightly](https://github.com/maelstrom2001/ArcticFox/wiki/Changelog)

Tomozaki123, 2017-03-14:
Yes, by ohm law. But on offical firmware i can set up to 200W, on specifications it's written that Pico Dual is capeble of 200W/50A?

Tomozaki123, 2017-03-14:
I update firmware, now it work's. Thank you for help. You are doing good work.

closed 2017-03-14 by TBXin