maelstrom2001/ArcticFox #215: [Enhancement] Store usage statistics

StillVoidingWarranties, 2017-03-13:
I guess the supported devices have some kind of storage which could be used to record usage data. It would be nice to record puff count, puff duration and average puff power, or even better the integral of power over time, also known as "energy". To measure this, the mod would have to sample the used power during puffs, average the values and store this average.

With this data the capacity of the batteries could be determined and even discharge curves could be generated automatically without relying on voltage alone.

Also, a timestamp could be stored with each puff to create usage statistics to answer questions like "How much do I vape?", "At which time of the day/week do I vape most?", etc.

wulfgrim, 2017-03-13:
Are you asking for something like this?
[image]

StillVoidingWarranties, 2017-03-13:
Yes, exactly, ideally with the possibilty to view the statistics on the device.

TBXin, 2017-03-14:
Unfortunately there are almost no free space in the ROM to keep statistics.
ROM size in our devices is 128kb.

At this moment we have a space allocation in the ROM like this:
0x00000 APROM (Firmware)
0x19000 LOGO
0х19800 EMPTY
0x1C000 SETTINGS (dataflashes, settings, curves, statistics, etc)
0x20000

You can noticed that firmware takes 100kb and this is almost all space.
It is also worth to remember the specifics of flash memory:
1. The memory is divided into pages of 2KB
2. Each page has a finite number of rewrites
3. Data is erased page by page

To increase lifetime of the ROM we are trying to reduce amout of the write operations.

At this moment we are collecting only two statistics items - puffs count and total puffs duration.
Each of the counter have own page in the ROM to keep their data.

Summing up and taking into account the limitations of flash memory, we will need at least a page of memory for each counter. 

We don't have such free space.

StillVoidingWarranties, 2017-03-14:
Ok, thank you, TBXin for your explanation.

Wouldn't it then be good to have a possibility to deactivate the puff counter altogether? I know many users do not care at all how many puffs they have taken. How would the firmware react if one of the puff counter pages would become inaccessible?

TBXin, 2017-03-14:
> Wouldn't it then be good to have a possibility to deactivate the puff counter altogether?

No reason to do this. 4 pages for counters are isolated and not using for something else.

> How would the firmware react if one of the puff counter pages would become inaccessible?

Nothing will happen. Just counters will have zero value.

closed 2017-03-20 by TBXin