maelstrom2001/ArcticFox #217: RX300 USB charging will not switch off after complete

RX300N, 2017-03-14:
IStick 100 and 200 work fine however RX300 does not switch off will even after reaching 100% battery level. With generic battery profile it will charge the batteries up-to cca 4,15V, then charging will stop (temperature goes down from app. 47C to 39c) however mod still displays charging icon. I tries it several times - even overnight but to no avail. Bug? Or am I doing something wrong? Does the same thing whether switched off or in sleep mode only. As I mentioned both Istick100 and 200 work OK.

maelstrom2001, 2017-03-14:
The last stage of batteries charging in RX300 is balancing. It may take a long time especially if batteries voltage measuring is incorrect. Take a look in Settings -> Expert -> BVO menu, normally difference between voltages should be about 0.03..0.05 volts. The best way to calibrate these measurements is to charge batteries in external charger, make sure that all voltages are equal, put batteries in RX300 and set correct voltages in BVO menu.
It seems that RX300 read batteries slightly incorrect and tries to balance it for a long time.
Also, note that 100% in GEN battery profile is equal to 4.10 volts.

RX300N, 2017-03-14:
Thanks - yes I know that, batteries were charged externally - then measured digitally and then calibrated with BVO to the same value. Up-to to here everything is correctly set. Batteries were then used for some 50-60 puffs at 70W and then put on USB charging. This ended with 4.12 voltage (displayed value) for all of them (balanced), onboard temp dropped from app 47C to 39 or so and charge level was at 100%. To my knowledge nothing else had to be done. However even with continued charging overnight (app 8 hrs) it was still in charge mode, with display not switched off. I think it was not charging any more (as per dropped down temperature) but the display should switch off - or does it work differently? Seems not to be a big issue - however you know, switching the display off after reaching 100% will be better.

maelstrom2001, 2017-03-14:
Please check option in Settings -> Expert -> USB -> NoSlp. If it set On, switch it to Off and try to charge again. Will take a look also in code, maybe really missed something :)

RX300N, 2017-03-14:
Thanks for a tip. I did two attempts **first with mod completely switched off** and then USB charged and second with NoSlp=On - the result was same. 
I assume that switching NoSlp to OFF should result to the same result as completely switching the mod OFF (5clicks) am I right? I will try your suggestion later on. Thanks so far

maelstrom2001, 2017-03-15:
Experimental build, try it if you want. Perhaps fixed some strange cases while balancing at the end of charging. Requires [nightly NToolbox](https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts).

[af_170315.zip](https://github.com/maelstrom2001/ArcticFox/files/843972/af_170315.zip)

RX300N, 2017-03-15:
Thanks for new nightly build. Re yesterday and 170201 build with NoSlp=OFF - there was no change compared to previous results. However one thing surprised (maybe it will be of some use) even getting to 4.15 voltage (4.1 should be the limit for USB charging with Generic battery) the charging apparently went on with a very slow pace. I left it overnight and after another 8 hrs it raises to 4.17 and adding another 6 hours of charge it raised to 4,18-4,19 and still charging was not switched off. 
Now I downloaded new build as per your instructions started new USB charge from 4.04V (measured by dig. voltmeter and approved by display on mod). I will come back within 1hr or two when the display show 100% - I am wondering what will happen. :-)

maelstrom2001, 2017-03-15:
Great, thanks in advance for testing and response!
Among other things, a little bug was found: charging current in last stage was set to 300 mA vs 600 mA in stock firmware. Have no idea why nobody noticed that :) It fixed now in the build above, perhaps it was one of the reasons why charging was so sloooooow :)

RX300N, 2017-03-15:
After 1hr 3batteries are flickering slowly between 4,11/4,12V last jumped to 4.11 temperature is 49c and battery percentage is now 100%, charging still continues.

RX300N, 2017-03-15:
After 2hrs 4.16/4.16/4.16/4.16(4,17 flickering) temp dropped within last 10mins to 43C, charging continues despite limit of 4.1V

maelstrom2001, 2017-03-15:
Sounds promising :+1: thank you! Crossing fingers, waiting for charge stop at 4.19 volts :)

RX300N, 2017-03-15:
3 hrs 4.22/4.22/4.22/4.21 temp 42C - charging continues. I am starting feeling unsafe - should I leave it charging overnight? Just for testing result? Is it safe? I will give it one more hour - then I have to decide what to do if it still does not stop charging.

maelstrom2001, 2017-03-15:
Charging up to 4.25 volts is safe. If charging will not stop itself in 1 hr or at 4.23 volts on any cell, please power it off manually.

RX300N, 2017-03-15:
4 hrs and I am giving up. 4.29/4.29/4.29/4.27 - displayed and measured by voltmeter. I switched it off manually. I have a bad feeling that threshold level of switching off is hardwired and cannot be influenced by software means. But I am no way an expert. I just remember that first try with original firmware finished quite unbalanced with one battery at 4.3 and the rest around 4.27 or so - which surprised me quite a lot. However with original firmware battery display shows different battery levels even when they are charged to the same voltage - and it cannot be corrected.
Thanks for assistance for today.

maelstrom2001, 2017-03-16:
Thank you very much for testing!

>  I just remember that first try with original firmware finished quite unbalanced with one battery at 4.3 and the rest around 4.27

"Balanced" condition in original firmware is means that voltage difference between batteries is not exceed 0.05 volts. But 4.30 volts cut-off is pretty high and it's not safe, I agree.
I think that charging routine in 170315 build is the same as in stock firmware (at least, I've try to copy it 1:1), but I see that it should be modified to be more safe.
Again, thanks a lot! If it is not burdensome, after a while I will ask you to test another build :)

RX300N, 2017-03-16:
One very important discovery and last for today 
I made some puffs, battery level dropped to 4.23, I started recharging and hey, it does not charge any more. So made another puffs battery dropped to 4.18 percentage level showed 98% and upon connecting to charger it started charging. Reached 100% and continued so I did stop it manually. Disconnecting charger and reconnecting again and voila charging was not initiated. So problem is somewhere here. Hope it helps.

maelstrom2001, 2017-03-16:
Thank you!
Charging routine was reverted to 170222 with 600 mA fix, hope it will work more correctly than previous...
[af_170316.zip](https://github.com/maelstrom2001/ArcticFox/files/847954/af_170316.zip)

RX300N, 2017-03-16:
Well done - congrats. 👍 🥇 
This build switched the mod to OFF when reaching 4.2V (100% with HG2 batt. profile).
One question though - it seems to me that charging current is somewhat limited compared to 170315. Board temp when charging is 43C - with 170315 it was 49C. Well higher charging current will be better - if possible. Anyhow you guys are SMART.

maelstrom2001, 2017-03-16:
Thank you very much for response! We have no new generation mods with balancing charger, so help to testing the code for them is highly appreciated.
Tell, please, what battery voltage was when you start charging on 170316? If it was >= 4.12, then charging current should be about 600 mA from the beginning. And yesterday you put mod on charge with 4.04 volts, in this case charging current should be about 1 amp. Both builds has 600 mA limit for the latest charging stage (balancing), they are different in stop charge cases only. The last and the longest stage of charging (balancing) starting when battery voltages are reached about 4.16 volts.
The last test to make sure everything is working correctly is to start 170316 with batteries about 4.00 volts, this will force the procedure to go through all the steps from the beginning to the balancing and switching mod off.

RX300N, 2017-03-16:
Yes my fault - I did not write down the start voltage for today´s charging. I think it was roughly 4.14 - and when so 600mA limit will made the device cooler for sure.
Now give me please some time to find proper battery set with this lower battery level. I´ll come back then.
Have a nice evening

RX300N, 2017-03-17:
Ok I made complete test in the evening and I can confirm everything running smoothly.
Started with 3 LGhg2 at 3.96 and one Samsung R25 at 3,86. You´ll find complete report here 
https://drive.google.com/open?id=0B0KRWC-En4-Reno3TDBKMVpHTjQ
It stopped after 4 hours with near perfect balanced batteries. Last hour was mainly balancing batteries - even though it was not necessary. So 98% capacity after 3hrs of charging and you are ready to go. Leaving the mod charging overnight seems to be safe. Questionable are temperatures which topped at 56C for some time - I suppose those are board temps - the mod itself was slightly warm but not too much. It is probably good idea to make sure that air ventilation holes are not blocked. If you have any questions do not hesitate to ask :-)

maelstrom2001, 2017-03-17:
Great, thank you very much! :+1: 

> Leaving the mod charging overnight seems to be safe.

It was the main goal :)

> Questionable are temperatures which topped at 56C for some time

RX300 using internal MCU temperature sensor. I don't know how close MCU is placed to high power dissipation parts and how precise it can measure temperature of charger circuits for example, but can say the one thing: when MCU is running, its temperature measured by this sensor can be up to 40 C, despite all other components and mod's case has room temperature. Running MCU is heating itself. So we can assume a slightly heightened temperature as the norm IMHO.

RX300N, 2017-03-17:
Well done guys 👍 , I close this thread

RX300N, 2017-03-17:
> Odesláno: 17. 3. 2017 14:50:00
> switch off after complete (#217)

Hi - now just one more question - because I have closed the thread.

Is this 316 build usable generally? I mean I have RX300 RX2/3 and then 
Istick 200 - can I use this version with those mods? Just trying to 
avoid the mess which build is tested for which mod.
Thanks - and have a great weekend

Vlad

------ Původní zpráva ------

maelstrom2001, 2017-03-17:
Yes, 170316 has support for all devices, and it is fully compatible with nightly NToolbox. You can use it on any mod without restrictions.
Have a nice evening :)

maelstrom2001, 2017-03-25:
@RX300N Hello! Can you test charging on attached build please? We made some changes for eVic Primo (https://github.com/maelstrom2001/ArcticFox/issues/106#issuecomment-289123277), now have to make sure that nothing is broken :) 
Thanks a lot in advance!

[af_170324.zip](https://github.com/maelstrom2001/ArcticFox/files/869926/af_170324.zip)

RX300N, 2017-03-25:
Ok no problem, will do when my battery set will drop to app. 4.0V (probably evening). 
Take care :-)

25. 3. 2017 napsal(a):
> @RX300N Hello! Can you test charging on attached build please? We made
> some changes for eVic Primo
> (https://github.com/maelstrom2001/ArcticFox/issues/106#issuecomment-289123277),
> now have to make [...]

-- Odesláno z telefonu s Androidem pomocí pošty K-@ Mail.

RX300N, 2017-03-25:
170324 does not work properly for RX300. Some measured values (v170316 in brackets)Temperature during charging maxxed at 62C(56c) - and the mod was pretty hot. In balancing regime temperature was 55C(44) final volatage reached after 3,5 hrs of charging was 4,12 (4,20). Starting voltage was 4,3x3+3,8x1 (3,96x3+3,86x1). To put it simply - significantly more heating/slower charging and final cut-off will probably not happen - within last hour voltage on 3 channels reached 4,12V and stopped there.
I have to upload  316 - works well.

maelstrom2001, 2017-03-25:
Thanks a lot! I think using 170316 on RX300 is the safest at this moment, so have to investigate what goes wrong exactly on Primo. But it is another story :)
Again, many thanks for help and detailed reports! :+1:

Ustes, 2017-03-27:
@maelstrom2001 I have an issue that happened while charging with USB to where now the battery indicator says balancing.  When I go into bvo battery #3 has 0.00 but #4 has 6.4.  I have charged the batteries on an external charger to 4.2v.  I apologize if this is the wrong place.

maelstrom2001, 2017-03-27:
@Ustes probably it's hardware failure, box does not see 3rd cell. Check balancer's contacts:

[image]

Ustes, 2017-03-27:
@maelstrom2001 ah, thanks..I played with them and now it is working..should they be flush to the battery door or have some space? or does that even matter, as long as they are touching the little plugs.

maelstrom2001, 2017-03-27:
@Ustes it's spring loaded pins, right? Maybe one of them has stuck slightly and do not moved out completely to assure a good contact with strip on battery door. Just check they are straight, moves well and touches strips when you closing the lid.

closed 2017-03-29 by maelstrom2001