maelstrom2001/ArcticFox #222: Eleaf Pico Dual Cant set more watts then 93.8

GFSNinja, 2017-03-20:
I have this problem with my pico dual that when i instal arctic fox i cant set more watts then 93.8 i dont know why but when i instal firmware its says Unknown device. 
Please reply fast i realy want my 200 watts with arctic fox.

maelstrom2001, 2017-03-20:
Update firmware and NToolbox to [the latest nightly](https://github.com/maelstrom2001/ArcticFox/wiki/Changelog).

GFSNinja, 2017-03-20:
i have
[image]
[image]

GFSNinja, 2017-03-20:
I have
[image]
[image]

GFSNinja, 2017-03-20:
Ty man you are the best.

closed 2017-03-20 by maelstrom2001