maelstrom2001/ArcticFox #223: Support for Pioneer4you IPV8

andreidotnet, 2017-03-21:
Hello guys,
Just a quick question - do you plan to have support for Pioneer4you IPV8 or any of the IPV series ?
[http://www.pioneer4you.com/goods.php?id=137](http://www.pioneer4you.com/goods.php?id=137)
Thanks and keep up the good work :thumbsup: awesome firmware

ctrlaltca, 2017-03-21:
All the actually supported devices uses the same Nuvoton chipset with minor changes (display size, clock, ..).
The Pioneer4you boxes are using a completely different chipset, so i'm afraid you'll get the same negative answer as #40 and #194

maelstrom2001, 2017-03-21:
Exactly.

andreidotnet, 2017-03-21:
Thanks for the reply.
Might be helpful to mention the chipset compatibility restriction/limitation at the top of the device compatibility list, in bold :) so others won't repeat the same questions.

closed 2017-03-21 by andreidotnet

maelstrom2001, 2017-03-21:
@andreidotnet or we'll add to the "for Joyetech, Wismec and Eleaf" word "ONLY" in huge letters :)
Other manufacturers - it's good, but for now we'll please you with new features for boxes of the "big three".