maelstrom2001/ArcticFox #224: How to install the FW on Joyetech Primo?

javak810i, 2017-03-21:
Hey, I wanted to install the latest 222 release on my Joytech Primo with 4.05 stock FW via NFE toolbox 1.5 and I'm getting the "FW not supported for this device" error. Any help?

ZenFree, 2017-03-21:
Did you try to install 170222 firmware and latest NFE build?
https://github.com/maelstrom2001/ArcticFox/wiki/Changelog
https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts

javak810i, 2017-03-21:
No I did it with 6.5.0
should the one you posted work?

ZenFree, 2017-03-21:
I don't have Primo to test, but new NFE version may help based on the error you have received. They just recently added support for Primo which wasn't available in previous versions.

closed 2017-03-21 by maelstrom2001