maelstrom2001/ArcticFox #229: Can't work out how to get + and - buttons back to the correct orientation on dual after flipping the screen.

wackydude1234, 2017-03-24:
I flipped the screen on the pico dual by holding + & - when powered off, but now the - button increases wattage and the + decreases wattage, but i want the screen in this orientation, can i put the buttons back without having the screen flipped the other way? Build: 170201

wackydude1234, 2017-03-24:
also the mod will only go to 100w.

maelstrom2001, 2017-03-24:
Flipping screen by holding + and - buttons also swaps these buttons, this is by design. If you prefer to do not swap them, switch option in NToolbox:
[image]

To avoid 25 amps limit of output current use [ the latest nightlies](https://github.com/maelstrom2001/ArcticFox/wiki/Changelog) (firmware and NToolbox)

wackydude1234, 2017-03-24:
Ah thank you, also what nightly do i use? I tried to do the top one and it said it's not compatible

maelstrom2001, 2017-03-24:
> it said it's not compatible

For nightly builds you should use [nightly NToolbox](https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts)

wackydude1234, 2017-03-24:
Thank you, your time and reply is appreciated.

wackydude1234, 2017-03-24:
This is now solved thanks!

closed 2017-03-25 by maelstrom2001