maelstrom2001/ArcticFox #230: af_170222.bin Pico DUAL

wackydude1234, 2017-03-24:
Just wanted to let you guys know I put my dual down for about five minutes while it was turned on and when i went to vape it again it wouldn't respond or light up, so i hit the fire button 5 times and the screen went all glitchy and turned off, then i turned it back on by hitting the fire button 5 times and it came back to normal.

TBXin, 2017-03-25:
Hello,

> so i hit the fire button 5 times and the screen went all glitchy and turned off,

That's power off animation, like CRT monitors.

wackydude1234, 2017-03-25:
That's fine but it still did freeze on me haven't been able to repeat it since though

maelstrom2001, 2017-03-25:
@wackydude1234 occasionally I get hangs on wakeup on my Pico 75W, it happens very rarely, maybe once in week or two. It's totally unpredictable, but still trying to do something with it. Maybe it some kind of hardware issues. Thanks for report.

closed 2017-04-26 by TBXin