maelstrom2001/ArcticFox #232: REQ: Battery Calibration on multi-battery mods

FrankBKK, 2017-03-26:
I am not happy with the USB charging of AF mods on more then one battery (tested it on Primo, Cuboid200 and RX300) - the balancing at the end of the charging cycle is taking hours and usually without being able to charge fully. Also the BVO does not seem to bring proper results. (tested including the latest nightly build 170316)

Wouldn't it be possible to add an extra menu for charge calibration where you first charge your batteries externally, then put them into the mod, open the calibration menu and enter the current voltage measured by external charger and let AF determine the BVO and correct balancing parameters ?

RX300N, 2017-03-29:
From my experience with RX300 I can say that last stage of USB charging takes no more then 1 hour - and when using at least similar batteries it is not even necessary. From 50% to 98% (startup volatge difference app 0,15 v) will end with 0,03-0,04 difference before balancing even starts. 
Re calibration - BVO works perfectly OK for me - just as you described fully charge externally, let the batteries rest for i.e. 10mins then measure and put in mod. Now I use BVO settings to adjust each channel to measured values and that is it. With battery profile LG HG2 and USB charging they will charge exactly to 4,2-4,22 (measured externally) volts and then charging cuts-off.

FrankBKK, 2017-03-29:
That's sounds about right and as it should be - unfortunately not on my mods - RX300 with paired batteries takes several hours to charge from around 50% to 8x% - that's all I get, even when charging overnight. 

I tried it with various battery-set's and brands (Samsung, Panasonic, Sony, AWT) that all work well in other mods. They usually stuck at around 3.95V - I use 'generic' and no BVO but also tried dedicated battery profiles and accurate BVO's but without success. 

The RX300 is the worst, followed by RX3 and Primo - all of them have similar issues - that's why I assume the problem is with AF rather than the batteries. On original FW all of them work fine.

For now I only charge multi battery-mods with AF on an external charger.

RX300N, 2017-03-29:
That is strange. AFAIK generic battery profile will cut-off at 4.1 v - this is set as 100% in battery profile. LG HG2 is set to 4,2V. I do not know about other battery profiles. Secondly - which charger are you using - me LG quick charger from LG phone. 
Just look in thread "RX300 does not switch off..." somewhere in the issues list. AMong others there is quite lot of data how the development of 170316 continued. Maybe you might find interesting excell table produced on final test of this build where you can find the whole USB charging process documented through hours temperatures and reached voltages incl. final cut-off

maelstrom2001, 2017-04-01:
In next release we'll increase charging current for the last stage by 5%. At first glance, this is not very much, but believe, this accelerates charging noticeably.

RX300N, 2017-04-02:
Great news thanks. Do you already know which release number it will be?

FrankBKK, 2017-04-02:
I can confirm the latest version 170331 does the trick ! 

Charging on RX300 and Primo now works perfect !  Also all other devices the charging works fine.

However I realized that with this version on the Evic VTC mini the charging does not stop at 4.1V or 4.2V - I manually had to stop it at 4.22V

closed 2017-04-05 by FrankBKK