maelstrom2001/ArcticFox #235: HotCig R150 support? Willing to donate generously :)

dcb-ror, 2017-03-28:
Hey guys, the HotCig R150 is an excellent device but the firmware is terrible. There's not even a 'stealth' mod option, let alone other basic stuff like puff count or the ability to turn the mod off/on (just a 5-click lock) 

It would be AMAZING to say the least if we R150 owners could get some ArticFox action on this mod. Is this possible at all? :)

TBXin, 2017-03-28:
Hello,

Thansk for your suggestion, but we are not planned to create an ArcticFox firmware for other brands.
All supported devices by ArcticFox firmware based on the same chipset with a little difference (screen, shunt, ADC) and we can handle this changes, but if we are talking about others brands - need to do a lot of research (create firmware from scratch). This is a very complex task in which we are not interested at the moment.

Some related issues: https://github.com/maelstrom2001/ArcticFox/issues/223, https://github.com/maelstrom2001/ArcticFox/issues/194, https://github.com/maelstrom2001/ArcticFox/issues/40

FrankBKK, 2017-03-28:
Agree with the OP - how about a Kickstarter Project to fund this project and wet your appetite ?

dcb-ror, 2017-03-28:
TBXin - thanks so much for your response, I can understand where you are coming from and how complex the task may prove to be. You have already done great, innovative things in this space. But I think (and I'm sure you know this more than I do) there is still room for even larger disruption. 

I've been posting on the [ecig subreddit](https://www.reddit.com/r/electronic_cigarette/) about this. I feel like OS code is so widely available for different applications like vapeheads are missing out bigtime. What a game changer it would be if there was an OS firmware of this excellent quality that was available for most modern and popular devices out there. Nothing against DNA, they are great guy and do great work. However, the spirit of OS is that we can all come together and benifit. 

If such a firmware were ever to be created, it would make a lot of sense of device manufacturers to fork the code and provide support for their own devices. Someone pointed out on reddit that this would take the attention away from firmware and force manufacturers to focus their design and marketing effors on the hardware itself. Which they might not desire. However, I feel this is already the case with all devices using a DNA chipset, which are very popular, and overtime it would be the companies who did not provide support for the 'best OS vaping firmware" that would begin to lose out, because they're homecooked firmware would be subpar to the rest of the field. 

If I'm confused or missing something please let me know. And count me in on supporting a Kickstarter to evolve the vaping game as we know it!

ETPNV13, 2017-04-16:
No offense but if the r150 was so "great" it would have firmware to match it. Writing paragraphs to the devs over a topic they already said they have no interest in and trying to garner support on reddit doesn't change the devs outlook when they have enough to deal with on their current project as it is...I am sure if enough people thought that the r150 was great then somebody else would make custom firmware for it

adzam016, 2017-04-24:
Once again, I agree with ETPNV13. Eleaf, Joyetech and Wismec are some of the best devices going. They have made the firmware for these as they are popular and relatively easy to build the firmware for as there are three companies on one chipset.

Perhaps you need to buy an Eleaf, Joyetech or Wismec mod. I'm in camp Eleaf all the way. I have all Eleaf mods and even the standard firmware on them is great. Very smooth and simple. Arctic Fox just gives them that little bit of extra control and some nice nerdy coil building features.

Go and buy an Eleaf and chuck your R150 in the bin!

closed 2017-04-27 by maelstrom2001