maelstrom2001/ArcticFox #239: VaporShark SwitchBox RX

MicheleLynnBass, 2017-03-29:
Please make a patch for the VaporShark SwitchBox RX, it is the same board as the VaporFlask Lite which works very well with ArticFox.  I can even get the native VaporFlask Lite firmware to work on it with the editor; however, I cannot get any of the ArticFox versions to work.  They all appear to upload but then do not work at all...no screen display, nothing...so I have to put back on the VaporFlask Lite firmware since VaporShark doesn't have any firmware available yet.

Thanks much for all your hard work!!!

TBXin, 2017-03-29:
Hello,

> VaporShark doesn't have any firmware available yet.

Without OFW for SwitchBox RX we can't add support to the ArcticFox.

We can only wait if sometime Vaporshark will release OFW for this device.
Also, seems to be this http://www.vaporshark.com/hedron is RX200S in another case.

MicheleLynnBass, 2017-03-29:
I'm using NFE v6.5 and I load the VaporFlask Lite 2.0 Firmware, then use NFE's Firmware Updater but hold down the control key while pressing update.  From that point on, it recognized it as a VaporFlask Lite.

TBXin, 2017-03-29:
@MicheleLynnBass holding down only "control" key does not affect anything.

I guess if you try to connect device in the safe mode to PC you will see "Unknown Device".

To enter safe mode you should:
1. Remove batteries;
2. Press and hold "Plus" button;
3. Connect to PC;
4. Release "Plus" button;

MicheleLynnBass, 2017-03-29:
I'm on a Mac running Parallels, so I'm holding down the right option key (which I assumed is control on a PC). So not really sure how the keys relate from MAC to PC.  The Wismec updater will not recognize the SwitchBox neither before or after trying to update to ArticFox.  Once you upload the ArticFox firmware onto it (by forcing it with the option key) then the mod doesn't work at all, so when you go to put the VaporFlask Lite Firmware onto it...you can't use the Wismec Updater because it won't update it and I can't force with Parallels. So I have to the NFE which doesn't recognize it either however I can get it to load the VaporFlask Lite at which point, NFE recognizes it as a VaporFlask Lite.  I then tried to upload the ArticFox firmware again at that point; however, same results...nothing works.  I'm going to try your recommendation re: safe mode.

maelstrom2001, 2017-03-29:
@MicheleLynnBass connect box in safe mode as **TBXin** wrote above (without battery and holding + button), start NToolbox Updater, read dataflash and attach .dataflash file that you'll get here.
You should get "Unknown Device" and firmware version 0.00 in NToolbox Updater, something like this:

[image]

MicheleLynnBass, 2017-03-29:
I'm not sure if this worked...keep getting a time out error.
[Unknown device HW v1.01 FW v0.00.zip](https://github.com/maelstrom2001/ArcticFox/files/880615/Unknown.device.HW.v1.01.FW.v0.00.zip)

maelstrom2001, 2017-03-30:
Yep, something goes wrong. Please download latest [NToolbox nightly](https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts), connect again in safe mode and post screenshot of NToolbox. It should show ProductID in square brackets before "Unknown Device".

MicheleLynnBass, 2017-03-31:
Downloaded the nightly build, still same problem however. 

[image]
[image]

maelstrom2001, 2017-03-31:
@MicheleLynnBass try to flash attached build :) nevermind "Unknown device", flash with NToolbox as usual.

[af_170331.zip](https://github.com/maelstrom2001/ArcticFox/files/885170/af_170331.zip)

MicheleLynnBass, 2017-03-31:
Works perfectly now!!! Thank you so much!!!!

MicheleLynnBass, 2017-04-01:
@maelstrom2001 one thing I noticed now that it is working, is that the LED light around the fire button no longer lights up when fired.  Still, the mod is so much better with the ArticFox firmware; however, maybe in a future version of the firmware you could put a command in to light up the fire button when pressed.

maelstrom2001, 2017-04-01:
@MicheleLynnBass if you'll find the original firmware it can be implemented. Without stock firmware it is impossible to find which port is controlling LED :)

MicheleLynnBass, 2017-04-01:
@maelstrom2001 I'll keep an eye out for it and post it here when I find it or they release it.  Thanks much!

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001