maelstrom2001/ArcticFox #241: Add 4th line option to small screen and 7th line option to large screen?

1881hc, 2017-03-31:
The right hand side of a pico small screen allows for one line of large text.  Could there be an option to have two lines displayed in that space?  i would use that option to display more information.  Could the same be done for the top line of the large screen?  I would take advantage of that too if it were possible.

Thanks for the good work on this excellent firmware.

P.S.  Make an android version for the play store and charge a few bucks.  I'd be a customer.

mclc, 2017-04-13:
I support this. It would be great to be able to have more information on the small screen.

Manych23, 2017-05-02:
Yes, that would be great. Also, it would be PERFECT, if author will make a constructor, where you'll be able to redesign the lines (for example, to change orientation to vertical on pico)