maelstrom2001/ArcticFox #242: [enhancement] on-device power curve editor

360up, 2017-03-31:
It would be nice if there was the ability to edit curves on the device, like on myevic fw

360up, 2017-04-01:
It doesn't need to be a full blown editor for pre-installed curves, but a 'custom curve' entry where you can enter 'time/pct' couples would be very useful.

Although I assume the effort to implement this would be quite the same.

TBXin, 2017-04-01:
Due to the small remaining free space in the ROM this feature most likely will never be realized.
Also, our approach is use configuration tool NToolbox to get maximum customization.
May be this is sad news for you, but this is our reality ^^ We choose to add support for more devices than add useless (in our opinion) curve editor.

360up, 2017-04-01:
> Also, our approach is use configuration tool NToolbox to get maximum customization.
May be this is a sad news for you, but this is our reality

While the tool box provides great usability -  especially for a complete pre-configuration directly after flashing - I'd like to point out that making features 'software-exclusive' limits usability.

> Due to the small remaining free space in the ROM this feature most likely will never be realized.

A pity, but it's all about compromises then anyways. For sure software curves are better then no curves at all :)

maelstrom2001, 2017-04-02:
>  I'd like to point out that making features 'software-exclusive' limits usability

It's a matter of taste, a point of view etc. DNA and YiHi boards has a limited number of option in box itself, but has pretty powerful PC software for fine tuning. However they are very popular and, I think, are very handy despite all that :)

closed 2017-04-13 by TBXin