maelstrom2001/ArcticFox #248: Rx200s battery

pikachukaki, 2017-04-02:
Im using the latest release and i have 3 lg hg2 batteries in the mod.
I use the hg2 profile in battery and enable battery percentance with the windows app but the indicator show 80% from the 1st puff.

maelstrom2001, 2017-04-02:
HG2 profile:

[image]

GEN profile:

[image]

I think battery voltage drops to about 4,02..4,05 volts after puff, and you can see percentage difference between profiles. If you want to have precise % calculation you have to make your own profile, 'cause it will be depended on number of battery cycles, your vaping power, resistance of atomizer etc.

pikachukaki, 2017-04-02:
I found some profiles and edit them to be imported to your app
[image]
although the values had to be edited cause your using only two decimals

docdeath, 2017-04-02:
where do i find discharge profiles  in arcticfox ?

pikachukaki, 2017-04-02:
In windows app In advance change the battery profile to custom and then edit them.

docdeath, 2017-04-02:
thanks did not know it was hiding there  lol

pikachukaki, 2017-04-02:
Me neither but when i saw the pictures i have to look for it!

closed 2017-04-07 by TBXin