maelstrom2001/ArcticFox #256: Wismec Predator 228 Support

zDeadKings, 2017-04-05:
Being these use the same chip, I almost assume AirticFox should work right outta the box. But before I brick my brand new mod would like conformation. Thanks for a great piece of work. Breathed new life into 1 of my fav mod the Cuboid.

TBXin, 2017-04-05:
#255 #252 #250 #249 #246 #234 #185

[image]

zDeadKings, 2017-04-05:
Thanks did not see it was a work in progress. I assume you will post  when it is available correct?

TBXin, 2017-04-06:
Let it be open. I hope people will not create duplicates ^^

leddhedd, 2017-04-06:
is there an ETA on the predator build? are we looking at days or weeks or longer? just so i can keep my eye out :)

TBXin, 2017-04-06:
@leddhedd ETA is a next release ^^
We will publish release version as soon as possible, there are no specific deadlines.

hunterj12, 2017-04-08:
How's the temperature control with stainless steel on the predator + arctic fox vs an alien? I'm trying to find  a new mod that actually works well and I would love to get a predator.

maelstrom2001, 2017-04-08:
@hunterj12 I recommend to watch [Dirk Oberhaus review](https://youtu.be/XXh3PrihtoU?t=31m41s) on RX300 with ArcticFox. On any other box TC will work the same way, so you can draw conclusions.

javak810i, 2017-04-10:
Any news on the Arctic fox FW for the new Wismec Predator? Would really appreciate that ;)

TheSpiritof69, 2017-04-10:
@javak810i https://github.com/maelstrom2001/ArcticFox/wiki/Changelog

javak810i, 2017-04-10:
@TheSpiritof69 Was rather looking for an answer like this "We're hoping to bring you Wismec Predator support in the upcoming 10 days or in the next release." Or something like this ;). But thanks anyway.

TheSpiritof69, 2017-04-10:
@javak810i Questions for ETA in non-commcerical, private software projects will in 99% of cases be answered with "It's done when it's done.".

maelstrom2001, 2017-04-10:
@javak810i In the next release :) with Eleaf QC support, powerbank mode (aka reverse charging for all devices), brand new skin and more. Please be patient :)

javak810i, 2017-04-10:
@maelstrom2001 Cool! Will be looking super forward to that :)

eruecco87, 2017-04-10:
+1 on the need of predator support, just bought one and found the Arctic Fox firmware.

Looks cool, great job guys!

eruecco87, 2017-04-13:
@maelstrom2001 I just uploaded the firmware to an Eleaf Aster, and I love it. This post is not to pressure a release but to let you guys know that Predator support will be highly appreciated.

Kudos again on the job

hunterj12, 2017-04-19:
Any chance the predator firmware will be released in April?

TheSpiritof69, 2017-04-19:
Everytime someone asks, the release will be pushed back 2 days. ⚡️

javak810i, 2017-04-20:
My Predator arrives on next week, so the Update will obviously be released next week :D
I wish :D.

leddhedd, 2017-04-22:
i can see that the readme has been updated to show support for the predator, is the new update imminent?

hawkeyexp, 2017-04-22:
Today it get my predator and noticed what good job the arctic firmware does :-) I Miss it :-)

javak810i, 2017-04-25:
Today I too joined the club of Predator owners :D. 
Super excited for the next release :)

mumpen, 2017-04-26:
Every day I come here and hope for the Predator support. 
You're doing a great job, hope we get the new version soon! 
<3

hawkeyexp, 2017-05-04:
Hi guys, can you give us some look inside the current state of dev ? Caused by the continious updates of upcoming readme it seems you make good progress...

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001