maelstrom2001/ArcticFox #257: RX300 400W output

maddpiratekidd, 2017-04-06:
Increase the max output for the RX300 to 400Watts as per official firmware.

ozomax1997, 2017-04-06:
would love this as ive upgraded to the new firmware but my screens are too dim as i have coloured films over the displays on both of my RX300's

semarx300, 2017-04-06:
rx300 400w please

TBXin, 2017-04-06:
No problem. We will update the maximum power for release version of the AF.

RX300N, 2017-04-06:
Yeah - even though it is practically nonsense, it would be great to have Ferrari with WIsmec logo LOL.

TBXin, 2017-04-06:
> it would be great to have Ferrari with WIsmec logo LOL.

and 1500 watts of output power ^^

docdeath, 2017-04-06:
400w lol really  going to do cheese  toasty on mine now  f##k the vapeing lol   rx300 owner as well

TheSpiritof69, 2017-04-06:
Guys, this is an issue tracker, not a forum.

Madad1996, 2017-04-09:
We need an english forum.

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001