maelstrom2001/ArcticFox #259: Latest builds

Dil3mm4, 2017-04-06:
Hello.
Istick Pico mega here.
The latest builds aren't working (msg: this build it's not for this device), the only one that currently works it's 170212.
Dunno if it's supposed to be like that, that's why I'm reporting back.
Have a nice day.

Tomozaki123, 2017-04-06:
Hello.
I have also Eleaf pico mega and latest build is working great. Try this one: [https://github.com/maelstrom2001/ArcticFox/wiki/Changelog](build)

maelstrom2001, 2017-04-06:
@Dil3mm4 Update NToolbox to the [recent nightly version](https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts).

Dil3mm4, 2017-04-08:
Solved, but @maelstrom2001, you could type, in the nightly release changelog, when there's a minimum version requirement for NFE.
It would be useful and will possibly avoid all the issues related to this matter. 
Hope this suggestion help, 
have a nice day.

closed 2017-04-08 by Dil3mm4