maelstrom2001/ArcticFox #264: Please port it to android

delchrys, 2017-04-10:
If this program goes to android it would be great. Everyone has his phone with hime so it would be easier to adjust and to monitor the vape statistics.

TBXin, 2017-04-10:
Hello,

1. We have no skills for creating Android applications;
2. NFE is an open-source project, so every one who interested in this with an Android apps development skills can create appropriate application;
3. If such a person exists, we will help and explain how to interact with the device;

delchrys, 2017-04-10:
1. Pitty
2. hope someone steps up, because I can't
3. would be great feature to have my mods talking to my Phone. So please someone....... help......... :)

sensitus, 2017-04-10:
and how do you suggest to connect them?

TBXin, 2017-04-10:
@sensitus root, OTG host, USB HID, profit ^^

sensitus, 2017-04-10:
I didn't see any microusb to microusb cable. Ever.

sensitus, 2017-04-10:
Oh, I see what you mean. Not worth the hassle imo

TBXin, 2017-04-10:
@sensitus 

Any Micro USB OTG Cable ^^

[image]

And yes, i agree with you, this is just fun & challenge and no more.

delchrys, 2017-04-10:
There are small micro usb organisation adapters. Put in your charger cable and done. I think it would be fun. And was maybe simple to do. But if you guys think it's too much hassle then don't

sensitus, 2017-04-10:
I'm tempted to hack together an ESP8266 adapter just for the fun of it though. Will probably do that as a hobby project after I'm done with the end exams in the winter hehe

vojaly, 2017-04-10:
I hope that the device is equipped with Bluetooth

TBXin, 2017-04-10:
@vojaly only one device from Joyetech/Wismec/Eleaf has Bluetooth and this is OCULAR C which we are not supporting, because this is a piece of shit ^^

vojaly, 2017-04-10:
Like Tesla's touch 150W? I own this box.

sensitus, 2017-04-10:
Obey the G Class. G Class is love. G Class is life.
.
.
.
.
.
qmini also

TBXin, 2017-04-10:
This devices are not supported by our firmware, so, why we are talking about them?

fonix232, 2017-04-10:
What makes the Ocular C so "utterly crap" that you won't support it?

I mean, okay, it comes with lots of unnecessary stuff. Nobody wants a gallery or an MP3 player built into their mod. But I don't see why nearly the only BT-enabled mod needs to be called "utterly crap" just because you don't like it.

TheSpiritof69, 2017-04-10:
@fonix232 Likely that the firmware cant easily be ported because I assume the Okular series uses entirely different chips.

closed 2017-04-13 by TBXin

eruecco87, 2017-04-17:
hey @TBXin you mentioned "If such a person exists, we will help and explain how to interact with the device"

Mind If I get that explanation? I might be able to hack something together.

TBXin, 2017-04-18:
@eruecco87 oookay,

To interact with devices you should use HID Protocol.
Here you can look how NFE products transfer data to/from device: https://github.com/TBXin/NFirmwareEditor/blob/master/src/NCore/USB/HidConnector.cs

Every action has own command with unique ID/Code.
Every command has own signature and you should sign your data before pushing in to device. You can look how this done in the CreateCommand method.

Maybe you want to learn something specific?

eruecco87, 2017-04-20:
@TBXin 

Thanks for the link, I'm trying to create a connection with an Ionic Framework plugin for mobile development, I understand that you're not familiar with mobile development but maybe you know the values of the parameters needed here to configure the serial connection: https://github.com/xseignard/cordovarduino#your-device-is-not-yet-known

```
let serialOptions: any = {
    baudRate: 9600,
    dataBits: 64,
    stopBits: 1,
    parity: 0,
    dtr: false,
    rts: false,
    sleepOnPause: true
};
```

TBXin, 2017-04-20:
@eruecco87 why you are asking about serial connection? Device communicates with PC via USB HID protocol, not COM (serial port)

int3ks, 2017-05-28:
https://nfeteam.org/forum/threads/android-tool-for-arctic-fox.51/  ;)