maelstrom2001/ArcticFox #265: Profile VTC5A

semarx300, 2017-04-10:
Please add profile discharge Sony VTC5A.

TBXin, 2017-04-10:
Battery discharge profile is a very private thing, we won't add new profiles or to maintain old ones. 
In the new release will not have preset profiles, will GEN and CUSTOM.

GEN is a generic profile, which are suitable for most type of batteries, and CUSTOM will give for an advanced users possibilities to setup own discharge profile.

https://github.com/maelstrom2001/ArcticFox/issues/206

RX300N, 2017-04-12:
AFAIK it ads nothing just a related information/linkage between battery voltage and calculated remaining power. It differs even for the same battery type dependent on battery life - older battery drops the voltage sooner then new one. However I might be wrong - as ususally :-)

TBXin, 2017-04-12:
@RX300N you are absolutely right!

closed 2017-04-13 by TBXin