maelstrom2001/ArcticFox #266: Possible Enhancement - C° to F°

MicheleLynnBass, 2017-04-11:
It would be a nice option/feature to be able to switch C° to F° for board temps.

TBXin, 2017-04-11:
Board temperature units depends on the selected units in the TC mode.

MicheleLynnBass, 2017-04-11:
To clarify, I mean on the "Info" screen.

TBXin, 2017-04-11:
Will fix this issue in the next release.

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001

MicheleLynnBass, 2017-05-09:
I just updated to the new version of the software and on the info screen I can't figure out how to change the C° to F°.  Please help.

TheSpiritof69, 2017-05-09:
Just scroll over your current range and scroll once more.

At 200°F press down once to reach 315°C and vice versa.

MicheleLynnBass, 2017-05-09:
That's for temperature control, I started this thread for the "Info" screen and requested an option to be able to set the board temp reading to either C° or F°.  This has nothing to do with temp control profiles.

TheSpiritof69, 2017-05-09:
Changing it in Temp Mode changes it for the Info screen as well.

MicheleLynnBass, 2017-05-09:
Ah...thanks so much!!!  Working now. :)