maelstrom2001/ArcticFox #269: Which mod´s will Firmware work on?

IngoPOW, 2017-04-16:
Is there a complete list on what the firmware will work?

I mean unlisted manufacturers that use the same pcb as the listed ones in the main documentation.

eruecco87, 2017-04-17:
The list is in the readme file https://github.com/maelstrom2001/ArcticFox/blob/master/README.md

If you mean unlisted manufacturers, then I guess unlisted answers the question, there is no list.

TBXin, 2017-04-18:
There are no more manufactures that are creating devices on the same PCB's.
All known devices is made by joyetech (wismec, eleaf), i'am sure that all boards are made by Joyetech ^^ and devices like Nobacco Apyron or Vaporflask Stout/Classic/Lite or Vaporshark SwitchBox RX / Hedron made by Joyetech but with own unique design.

closed 2017-04-24 by TBXin