maelstrom2001/ArcticFox #271: Edit TFR profile issue

Hatake044, 2017-04-19:
Hello team,
A little issue when trying to edit TFR profile (I enter values manually because I couldn't import csv files..) is that I can't set temperature lower than 0°F and resistance lower than 1.0000 so to get the TFR curve from steam engine fully loaded.
I'm not sure if this is hardware limitation (I use eleaf pico 75) or software/firmware's.
it just makes you feel the vape incomplete or unachieved so I hope you could create a fix to it.
(sorry for my english, not my native language)

maelstrom2001, 2017-04-19:
> I can't set temperature lower than 0°F

Do you locking your atomizer resistance at temperature below -17°C? О_о

TBXin, 2017-04-20:
@Hatake044 you can import curves from steam-engine (from old version site), new steam-engine version slightly changed exported cvs files format. NFE nightly builds already supporting new files or you can just remove fractional part of the temperature numbers to get import works with a release NFE version.

Hatake044, 2017-04-21:
@maelstrom2001 not when vaping, why would I do that lol
I meant when editing the TFR CURVE: temp can't be set under 0°F and resistance factor can't be set to a value lower than 1.0000. Now the issue for me is that TFR curve for most of the wires has its starting point at temp < 0°F and res factor < 1.0000.
If I bypass that starting point and enter the remaining values from TFR table then I'm left with the last point of curve/TFR table blank !
Here is an exemple of TFR table from steam-engine 
Temperature control results
TFR
Graph
Table
Equivalents
DNA200
Temp. °C	-50	20	100	150	200	250	300	427
Temp. °F	-58	68	212	302	392	482	572	800
Res. factor	0.92102	1.00000	1.07950	1.12580	1.16780	1.20750	1.24630	1.33376

maelstrom2001, 2017-04-21:
We don't use TFR lower than 1,0 (assumed that coil resistance will be locked by user at room temperature 68°F):

[image]

Hatake044, 2017-04-21:
Understood, the soft is still great even without that but I thought it would have better compatibilty with csv tables from steam engine, if making it with 8 temp points to adjust + res factor go below 1 and temp below 0. Also some TFR for specific wire brands can not be fully entered in the soft the way it is now, see for instance Thunderhead ss316 TFR : 
Temperature (degF)	Electrical Resistivity	
-100	0.8937020928280579	
0	0.9563181394617516	
70	1	
200	1.0716916737768711	
400	1.1700369730644635	
600	1.2556546590916973	
800	1.332292260628889

TBXin, 2017-04-22:
@Hatake044 

70 - 1 
200	- 1.0716
400	- 1.1700
600	- 1.2556
...
calculate manually two points using linear interpolation function.
eg:
670 - 1.2824
740 - 1.3092
...
800 - 1.3322

http://www.ajdesigner.com/phpinterpolation/linear_interpolation_equation.php

[image]

Hatake044, 2017-04-22:
@TBXin thank you :) this is a great solution for me.
Really appreciate your help

closed 2017-04-22 by Hatake044