maelstrom2001/ArcticFox #272: rx200s my pc is not reading it any more

sabbagh21, 2017-04-20:
every thing was fine i downloaded the firmware connected the rx200s to the pc and it was successfully installed,the problem is the next time i tried to reconnect  it again the pc doesn't recognize it !! it's only showing that it's charging on it's screen but nothing else..
ps: i tried two different laptops and different usb cables and still not working
i need help please

TBXin, 2017-04-20:
What kind of software did not recognize your device?
Did you hear typical sound when connecting device to the PC?
What firmware are you using?
Device still working fine(except USB) ?

sabbagh21, 2017-04-20:
windows 10, no no voices at all it's just showing that it is charging and yes it's working perfectly except for the usb

sabbagh21, 2017-04-20:
and the firmware is 170201

maelstrom2001, 2017-04-20:
Try to connect box to PC in recovery mode: remove batteries, press and hold plus button and plug USB in. Box should be detected by PC, if not - check USB and cables again.

sabbagh21, 2017-04-20:
Well I tried to do that but still not working
Could it be possible that the device's USB port defected?
Thanks in advance

TBXin, 2017-04-20:
Yes, probably this is a hardware issue.
Try gently to clean USB connector on the device. May be some kind of dust in.

sabbagh21, 2017-04-20:
Alright thanks guys

sabbagh21, 2017-04-29:
Hello there ..guys i tried everything different connectors cleaned the USB socket and nothing happens.At first i thought it was the USB port that is defected or broken so i went to an expert to fix it and reinstall a new one,he checked it and said that i can put you a new socket but i think that its not the USB socket's problem it's a software's problem!! by the way it sometimes reconnect to the pc but most of the time it doesn't and when it did i uploaded wismec's firmware and also that didn't help reconnecting to the pc. Guys please i need help really i did everything but nothing worked!!
Thanks in advance

maelstrom2001, 2017-04-29:
Hello! If your box is not detected by PC in recovery mode (without batteries and pressed + button while connecting) we can not help, 'cause it is hardware issue. In recovery mode box loading write protected part of ROM which is not modified by AF firmware.

sabbagh21, 2017-04-29:
thank you so much for the fast reply. but do you think if i change the usb socket it might work properly?

maelstrom2001, 2017-04-29:
Before replacing socket, check it, if it looks undamaged perhaps it will be enough to resolder it. If that does not fix problem, check tracks to MCU.

ZenFree, 2017-04-29:
@sabbagh21 You or your tech body can use Ohmmeter to check USB connection.

closed 2017-05-15 by TBXin