maelstrom2001/ArcticFox #273: eVic AIO temperature "joke"

Zlatorog, 2017-04-20:
Everything is working like a charm... :+1: , but...
On display, when I pull, I have rows:
1. Settings name (small)
2. big ... C/W
3. Volt: ... (instead of volt I have 3 seconds pull time)
4. Coil: ...
5. Last: temperature in C

At 5, most of the time temperature is 21 C, sometimes 95 C, couple of times even 128 C...

Interesting is, that eVic is working the same all the time, it's not warmer or hot...
Did I miss (mess) something?
Thanks in advance, Mat

ZenFree, 2017-04-20:
It shows you last temperature of your coil.

Zlatorog, 2017-04-20:
I know that, but:
- I pull 10 times and it shows 21 C
- next day, I pull once and it shows 128 C
- after couple of hours, it's 21 C again...
- and so on...

If it would be constant, all the time same temperature for the same coil, OK, but, up and down like roller coaster...

TheSpiritof69, 2017-04-20:
On what kind of coil and what mode is this?

Zlatorog, 2017-04-20:
Wire: VW
Coil: LVC Clapton 1,5MTL

I tried other wire settings: SS, 316, 316l, but it's not working at all...

TheSpiritof69, 2017-04-20:
Temperature Control works only on specific wires. If your wire is not temp
control compatible, like Kanthal or N80 the temp display will only show
garbage values and can be replaced, it would work on SS coils/wires, on
Titanium coils/wires and on Ni200 coils/wires

Zlatorog, 2017-04-20:
Thanks!!!
So, "no real harm done", whatever the display show...
Thanks again... :100:

closed 2017-04-21 by TBXin