maelstrom2001/ArcticFox #274: [BUG - HIGH PRIORITY] Temp control on SS316L coils

Dil3mm4, 2017-04-20:
Hello there. 
iStick pico Mega hardware. 
As stated in the subject, the TC doesn't work. 
If you set up a Temp of 200C° the maximum it reaches it's 190° and stays there even if you keep firing. 
Happens even choosing TCR modes with a range value from 88 to 92.
Hope that this can be fixed soon.

joeblowma, 2017-04-20:
When I can't reach my set temperature (at which point it will say 'protection' and take control of the wattage to keep that temperature) it's because I haven't set the max wattage for that profile high enough. Perhaps you haven't set your wattage high enough?

I have 2 clicks set to 'edit' so I can change it easier, otherwise I have to enter the menu, edit the profile, turn TDom off, exit and change it, then go back into the profile and put TDom on again. Or hook it to a computer and use the config tool to change it.

maelstrom2001, 2017-04-21:
I can assume 2 cases in addition to **joeblowma**'s comment:
* locked resistance is invalid;
* used notch coils, which are the pretty strange things from TC's point of view :)

closed 2017-04-26 by TBXin