maelstrom2001/ArcticFox #277: Istick pico charging

Manych23, 2017-04-21:
When I charge my pico, it charges for about 10 minutes, and then charging stops (but it shows, that battery isn't full), so I shoud disconnect and connect usb cable again.
I think, it's because of mod temperature warning. Can you make a setting, which change the temperature of mod, to stop charging.
On stock firmware, there aren't such problems.
Thank you

maelstrom2001, 2017-04-21:
> battery isn't full

Depending on selected battery discharge profile you may never have 100% level, except GEN.
Charging on single cell boxes can not be controlled by firmware. Probably, Pico can not charge battery up to 4,18..4,20 volts (my Pico stops charging at about 4,16 volts), but on stock firmware 100% is equal to 4,10 volts (the same for GEN battery profile), so you never faced this issue on genuine Eleaf firmware.

closed 2017-04-26 by TBXin

Manych23, 2017-05-02:
No, my pico, when charging from 5V/1A, stops charging at 3.8-3.9 volts. And I should connect it to 0.5A charger to continue. There isn't such problem in stock