maelstrom2001/ArcticFox #283: new version

sigaraci, 2017-04-22:
We have been waiting a long time. Can we find out exactly when the new version will come out? 🥇

TBXin, 2017-04-22:
asap

maelstrom2001, 2017-04-22:
https://github.com/maelstrom2001/ArcticFox/issues/256#issuecomment-292997417
https://github.com/maelstrom2001/ArcticFox/issues/256#issuecomment-295405448

ETPNV13, 2017-04-23:
since apparently no one can be bothered to read through the official forum like a person who is actually interested in getting answers to this question without pissing off the developers like a million times with a million of the same question they very specifically point out an estimate of "the end of April or possibly beginning of May" They do a lot of hard work to bring us this firmware, the least we could do is put in the hard work to look for the answers that are literally already there for us to find if we just look for them

ETPNV13, 2017-04-23:
@maelstrom2001 and @TBXin and everyone else who is working on this firmware for us, thank you for all your hard work and dedication to this project, you guys are awesome!

vojaly, 2017-04-23:
My hunch is that the next version will be important upgrades.
Thanks a lot.

closed 2017-04-26 by TBXin