maelstrom2001/ArcticFox #284: Charging Current / QuickCharge Mode

hawkeyexp, 2017-04-23:
How do you think about to show current charging current (in mA) and QuickCharge-Mode on newer devices ? I think would be nice to see if charge will go fast or slow - possible to identify if third parity adapters will work correct.

maelstrom2001, 2017-04-23:
> show current charging current (in mA)

Yep, it's already implemented and will be included to release version.

> QuickCharge-Mode

If you mean Qualcomm Quick Charge technology, sad to say, it is impossible on current devices architecture.

hawkeyexp, 2017-04-23:
If i understand it correct it will be possible to show current charging current but not if quickcharge is active and in which step ?

maelstrom2001, 2017-04-23:
Quick Charge technology requires input voltage higher than usual 5 V. Our devices is not support QC protocol for QC chargers, they are designed to be charged at 5 V.
New boxes can drain up to 2 amps from start.
Then current will be lowered to about 0,4..0,6 amps when battery voltage reached about 4.12 V, as batteries requires lower charging current at this stage.
Charging current will be displayed always during charging (on multi-cells boxes, for sure).

hawkeyexp, 2017-04-23:
Hm are you sure ? For example the recommended charger for the predator is the avatar quick charger and it supports quallcomms's qc2.0 - but possible it is as some times a marketing joke from asia :-)

OlafKroll, 2017-04-23:
Quick Charge 2.0 DC 5V/2A, 9V/2A, 12V/1.5A
Predator only 5V/2A

hawkeyexp, 2017-04-23:
ok :-) missunderstable by wismec :-)

closed 2017-04-23 by hawkeyexp