maelstrom2001/ArcticFox #286: eVic Primo 2.0 support

javak810i, 2017-04-24:
New, updated version of eVic Primo, with more power (228W now), better 510 connection, 2A charging and reverse charging. I think it would be nice to see the ArcticFox fw on this device as well, since Primo is so popular.

adzam016, 2017-04-24:
It won't happen unless it is on the same chipset as Eleaf, Joyetech and Wismec. The developers expressed no desire to support other mods yet as all their time is used on the current firmware.

javak810i, 2017-04-24:
@adzam016 well I'm 99.9% sure the chipset is completely the same as on the old Primo, so porting shouldn't be much demanding.

TheSpiritof69, 2017-04-24:
Considering the Primo 1 is already supported and I assume it's rather safe
to say the Primo 2.0 228W chip might be the same as the one on the 228W
predator it is likely the 2.0 will be supported at some point in the
future.

ZenFree, 2017-04-24:
Primo 2 appears to be similar to Predator. The Predator should be supported with the next release.

maelstrom2001, 2017-04-24:
https://github.com/maelstrom2001/ArcticFox/issues/256#issuecomment-292998166

adzam016, 2017-04-24:
Nice! Looks like you guys are getting it! Can't wait for the iStick QC to get added to. Great mod but the brightness is insanely low and ELeaf stock software is so basic lol

closed 2017-04-24 by TBXin