maelstrom2001/ArcticFox #287: Clicks not working w/ TC profiles

crearl85, 2017-04-24:
It seems that I cannot use the 2 click when using a profile set to TC.  if I change the atomizer then edit the click menus it does not stick, it seems that the click settings are unique to each profile so I cannot edit any of the settings I need to use on the TC profile because they are set to 2 clicks and I cannot use 2 clicks on the TC profiles, it just tries to fire the coil

TBXin, 2017-04-24:
Clicks are sharing between VW and TC profiles, if you setup 2 clicks as T.Dom - it will work for all TC profiles.

crearl85, 2017-04-24:
that would be great if 2 clicks would work in tc.  not sure why, but it does not work at all, only 3 and 4 clicks work on a TC profile

TBXin, 2017-04-24:
Because you are not assigning any action to the 2 clicks.

crearl85, 2017-04-24:
Ok I'll have to take a look once I get back to my PC.  Thank you, I'll be sure to report back.

closed 2017-05-15 by TBXin