maelstrom2001/ArcticFox #288: Profiles think different atomizer is the same

crearl85, 2017-04-24:
I build most atomizers around .2-.3
when I put one of them on for the first time it'll ask which profile to use so I assign that atty a profile.  when I put on another atty w/ a resistance within .5 ohms it thinks that the new atty was the one I originally assigned to that profile.  I have 3 attys that my mod recognizes as one.  is there a way to correct this?

TBXin, 2017-04-24:
Firmware is determines profile depending on the attys resistance. If you have few attys with a same resistance "Smart" is not for you.
Smart default range is 10%, depending on the fixed resistance in the profile.

e.g. you profile has 0.5 Ohm saved resistance. If you install attys with resistance 0.45 - 0.55 - firmware will chose your profile.

I can suggest to you turn on "Smart".

crearl85, 2017-04-24:
if I were to lower the range to 1% would it be more sensitive?  I've already tried to lower it with no luck, but is it supposed to be more sensitive when lower and less sensitive when set to the 15% max?

TBXin, 2017-04-24:
> if I were to lower the range to 1% would it be more sensitive?

Yes, it does, but this is too sensitive for our devices. I mean that resistance can float when you are screwing attys or when you are pressing Fire (coil became hot and material resistance increments).

> but is it supposed to be more sensitive when lower and less sensitive when set to the 15% max?

Yes, if you are increasing Smart range - this allows more attys to satisfy your profile.

closed 2017-05-09 by TBXin