maelstrom2001/ArcticFox #294: Fimrware update notify

hawkeyexp, 2017-04-26:
It would be great to see that nfe toolbox will notify about new firmware updates. Optional selectable for stable / nightlies.

adzam016, 2017-04-27:
Agreed. Just a simple mailing list that tells us when it is out.

TBXin, 2017-04-27:
Yeah, we think about this feature, but it is more complicated than it really is.
I'll try to explain:
Specific version of the NToolbox can work only with specific version of the ArcticFox (mostly i mean "Configuration").
Yes, we may have a situation where either NToolbox either ArcticFox updated and not break compatibility, but this situations is caused by minor updates or bugfixes. Major updates almost always require updates to both products. 
I still have no idea how to make such updates loyal to the user that nothing was broken.

amrdia, 2017-04-27:
u can make it notify for a new version of ntoolbox also

TBXin, 2017-04-27:
@amrdia it does not solve the problem still.

Case 1:
1. User receive notification about NToolbox update;
2. Install update
3. NToolbox can't connect to the device because firmware data exchange protocol is depricated;

Case 2:
1. User receive notification about firmware update;
2. Install update
3. NToolbox can't connect to the device because firmware data exchange protocol is not compatible with old version on the NToolbox;

ZenFree, 2017-04-27:
I would just make program notify client when updates are available and provide them a link to download. In the page you provided link to should have short description with the requirements for this update (like to install new firmware you have to update Ntoolbox to version X first).

TBXin, 2017-04-27:
@ZenFree you can subscribe to the mailing list about releases on GitHub.

hawkeyexp, 2017-04-27:
I think a simple info when new version is available should solve the problem - when anybody has the skill to use a custom firmware he should know what he is doing and understand that he has to upgrade the toolbox after new firmware version is flashed to all his mods.

ZenFree, 2017-04-27:
TBXin, you misunderstood me. Personally, I don't care. I check this site regularly anyway... I was just sharing ideas, trying to help.

smok-e, 2017-04-27:
> > Case 1:
> 
>  1.  User receive notification about NToolbox update;
>  2.  Install update;
>  3.  NToolbox can't connect to the device because firmware data exchange protocol is deprecated;
> 
> Case 2:
> 
>    1. User receive notification about firmware update;
>    2. Install update;
>    3. NToolbox can't connect to the device because firmware data exchange protocol is not     compatible with old version on the NToolbox;
> 
> I see only one solution, firmware must be supplied as part of NToolbox. This will guaranteed that both products are compatible.

in the first case
4. user gets a message: "NToolbox is not compatible with your firmware version, please revert your NToolbox or update your firmware"

in the second case:
4. user gets a message: "please update you NToolbox to recent version"

wouldn't that be possible? 

jm2c

TBXin, 2017-04-27:
@smok-e NToolbox already says that firmware version is incompatible or NToolbox is obsolete, but many users can't read this messages and create issues...

hawkeyexp, 2017-04-27:
This is the case where a minimalistic base communication makes sense to identify firmware version and inform the user that he should upgrade his firmware

ZenFree, 2017-04-28:
@TBXin Thank you for advise. I think, subscribe is a good option for people who visits developers sites, but for general population not so much.

TheSpiritof69, 2017-04-28:
@ZenFree The general population isn't flashing custom firmwares on their mods either.

amrdia, 2017-04-28:
that's not a custom firmware that should be released oem for god sake

adzam016, 2017-05-04:
Wouldn't an email Mail Merge/email distribution group suffice for this?

Subscribe box that sends emails to an Excel spreadsheet/.cps file from the website and then just send to contact group(s) on email client.

I may be wrong but I thought this may be quite easy. I know mail merge/distribution groups are very easy to set up and use but sending email addresses from the website "Subscribe" box, I have no idea!