maelstrom2001/ArcticFox #295: Evic Vtc mini rebooting after hit at max power

porkokan, 2017-04-27:
After firing an rda with full power (75w) in VW mode the device just reboots, it doesn't happen every hit but it's pretty common, looks like when the battery sags and the voltage drops under a certain level the device just shuts down. I'm using samsung 30q batteries which never were a problem in any device, even with the original firmware.

maelstrom2001, 2017-04-27:
Looks like the battery voltage drops faster than box can measure its voltage and stop fire or reduce output power. If you are using not GEN discharge profile, try to switch it to GEN, or create your profile based on GEN with higher cut-off and 0% values, something like that:
[image]

porkokan, 2017-04-27:
I've been using the 30q profile, i'll try upping the cut-off voltage and play with the discharge curve then let you know how it works, thanks for the fast reply =)

maelstrom2001, 2017-04-27:
As the owner of VTC Mini and VTwo Mini I can say that the first has high demands on discharge profile settings on high output power, because it has no additional hardware battery monitor (as its successor, VTwo Mini and more newer models) which can generate interrupt and stop fire immediately on it.
Also, IMHO 30Q is not the best cell for 75W vaping. Perhaps, it is worth paying attention to more high-current cells designed for continued current drain at least 20 amps.

traviswparker, 2017-04-28:
The stock firmware does this as well.

hawkeyexp, 2017-05-02:
Had the same effect on vtc5 at 65W on 0,120Ohm - solution is easy - reduce the first powerup with a preheat with lower power than final used in your setting (in my case 40W / 0.2s) - this reduces the first hard drop - but as maelstrom2001 told you - change the cell

closed 2017-05-09 by TBXin