maelstrom2001/ArcticFox #297: key lock

sigaraci, 2017-04-30:
Thank you for everything first priority. I have been using this application for a long time and I am very satisfied. There's only one thing I want you to do. I see that the keys can be locked when the fire key is pressed 5 times in the new update. Instead, press and hold the + and - keys for 1/5 second to lock all keys, including the fire key. Everyone will be happy with it. Nobody in the past will choose to lock the keys instead of turning off the device after 5 times the fire key. I am very happy to fulfill this request

TBXin, 2017-04-30:
Hello, 

In the next update you will able to choose what 5 clicks will do: turn off device or lock all keys.

sigaraci, 2017-04-30:
yes I know. But my request is not locking with 5 clicks. Press the + and - keys briefly to lock. Believe me this will be more useful. Think about it, please.

TBXin, 2017-04-30:
> + and - keys for 1/5 second

We will not reduce the delay interval to 1/5 of a second. Current is 1 second.
If you are tired to hold buttons for 1 seconds you may press 5 times faster than 1 second and got locked device.

> Everyone will be happy with it.

Doubtful.

sigaraci, 2017-04-30:
[Filmim.zip](https://github.com/maelstrom2001/ArcticFox/files/966766/Filmim.zip)
See for yourself how handy it is. There is video in the zip file. I mean this.

ETPNV13, 2017-04-30:
@sigaraci are you just trying to piss off the creators?

ETPNV13, 2017-04-30:
@sigaraci I mean this is like 8 "issues" in a week of you thinking that the people who put hard work into this firmware should just do what you want with no questions asked...can you just let them release the firmware update we are all waiting on without people pissing them off already?

sigaraci, 2017-04-30:
No we will use it. And I want to contribute to it.

ETPNV13, 2017-04-30:
Then figure out what tools you need to make it the way you want it without pestering people who give us this stuff for free

sigaraci, 2017-04-30:
Everyone has an idea and they are evaluated. You do not have to do this. If your ideas are worthless, this is your problem. Nobody will bother about it. On the contrary, they will be pleased to offer my opinion.

sigaraci, 2017-04-30:
If we do not use this software, there is no sign of this software. That's why we have the right to present an idea. It's their choice not to do. But if the software is made for us, our ideas are important.

ETPNV13, 2017-04-30:
There is a difference between contributing good ideas that can benefit the devices and asking for a firmware to be specifically tailored to just you, let's break down some of your "contributions", shall we?

font...you can edit the font to whatever you like using the tools they have already provided

bypass mode...simply set the device to whatever battery voltage you are looking for, done

resistance lock in vw mode...why would you ever need this? it is only useful for temperature control

reduce start-up logo screen time...it last for 2 seconds, how is that prohibiting you from using your device?

bothering about new version multiple times...either read the damn forum that they provide links to or read through the issue logs to see this has been asked a million times already

So how exactly are these thoughts "contributing"??? If you would just do some research before just blindly posting your "opinions" would be valued much more I think

ETPNV13, 2017-04-30:
Ideas's and input are valued when they are well thought out and present something useful

TBXin, 2017-04-30:
> But if the software is made for us, our ideas are important.

2 mistakes in one sentence.

First of all we are created this firmware for personal use and only after some time we decide to release it to public.

With what you have, what exactly are your ideas important?

We read users suggestions. We are trying to understand them. We are answer the questions. We don't get paid for it. This is not enough?

What makes you think that we should implement any feature from any user?

ArcticFox firmware is our vision how firmware should looks and feels.

ArionWT, 2017-04-30:
@sigaraci,I can't see other way to use so fast key lock, if you not always taking turned-on mod with you. But, just imagine the situation, when you put mod in your bag, and something pressed buttons at the same time by chance. This idea is really useless.

ZenFree, 2017-04-30:
@TBXin The software/firmware you made is VERY good, it competes with the best mods software out there and significantly better than original firmware. I see all around ecig forums (at least english forums) are talking about ArcticFox. You are getting recognition.  I understand that you made software for yourself but may be if you team up with Wismec or Joyetech and make joint mods (like Wismec Sinuous Predator mod) you can generate some profit that you can use for development?

Ideas are just that, if you feel they are useless or too much work for little benefits then of course ignore them :)

TBXin, 2017-04-30:
@ZenFree thank you for your kind words!

> if you team up with Wismec or Joyetech 
They are not interested in this. It is even at hand. They sell the device, not the software. We make their devices popular. People choose their products because of our firmware (i saw realy many people that never see OFW). They get more profit. Why would they share it with us? :)

ZenFree, 2017-04-30:
@TBXin You are right. The only reason I bought Predator is because of ArcticFox! I almost always bought DNA devices before and wouldn't even consider Wismec vanilla version. There are few hundred people like me who goes on internet and search for something new. I think, if Wismec can put extra features your software brings right on their retail box, use it for advertising then thousands more would want to buy their products.

ETPNV13, 2017-04-30:
@TBXin and @maelstrom2001 Yeah, not gonna lie...the firmware is amazing and I love the circle theme (can not wait to see the foxy theme lol) I am literally on my computer every night waiting for the next firmware release because I just can not bring myself to use the predator on stock firmware, using Arctic Fox has tainted me as I now hate the OFW that these devices come with and what you guys have created is so much better and is so much more developed in a way that you can actually see you guys care about what you have created. You put a lot of time and effort into this stuff and it shows in what you release. I like that you actually take pride in your work and I do not think you guys hear enough of how amazing it is. Thank you for releasing what you have made to us and letting us be a part of what you have created!

ZenFree, 2017-04-30:
@TBXin There are actually other possibilities even if Wismec/Joyetech are too foolish to cooperate. All you need is Joyetech chipset, which you should be able to buy separate.  You could team up with anyone who has hardware manufacture capabilities (SvoeMesto comes to mind, but any would do).

Astrologies, 2017-04-30:
@ETPNV13 Same, I recently purchased the Predator 228 and ever since ive been checking back here every day.

escoltajuverf, 2017-04-30:
@Astrologies  remember to be careful with the predator, 510 negative will come off, and also removing the batts when device is doing balance charging phase will kill the boost converter chip. Also the two metal bars are not sealed any juice leak will enter the mod.

Astrologies, 2017-04-30:
@

Astrologies, 2017-04-30:
@escoltajuverf I never take the tank off the mod, so far I haven't experienced any real bad side effects besides occasional fire button glitch

ETPNV13, 2017-05-01:
@escoltajuverf I already did a fix for the 510, I sanded a little with a Dremel right next to the 510 and just put a couple daps of solder to connect the body and the 510 since I always have a tank or rda on my mods, it's not like I will ever see it and I never have to worry about it coming off unless the solder breaks which should take a lot more effort than just pulling out the 510 lol

closed 2017-05-08 by maelstrom2001