maelstrom2001/ArcticFox #298: iStick 100W compatibility possible?

drachenbaby, 2017-04-30:
First of all thank you for your awesome work!
The Arctic Fox firmware has made my devices soooo much better than before!!

But I have one question:
Is it possible to make Arctic Fox available for the iStick 100W?
Since it is alread available for the iStick TC 100W i guess it shouldn't be very complicated.
I know that the iStick 100W isn't TC capable but other features like preheat would be very usefull.
And since the iStick 100W is available extremely cheap, upgraded with Arctic Fox, would make it an awesome Box for the TPD2 Stock.
Thanks in Advance!

TBXin, 2017-04-30:
Hello,

iStick100W has no way of firmware updates via USB cable.
Therefore we can't install an ArcticFox firmware.

drachenbaby, 2017-04-30:
Ah okay.. didn't know that.
Thank you for your quick response and thanks again for your awesome work!!!! :)

TBXin, 2017-04-30:
@drachenbaby thank you too :) Have a nice day!

closed 2017-04-30 by TBXin