maelstrom2001/ArcticFox #299: Just in case anybody else wants it...

ETPNV13, 2017-05-01:
Just something I made for my PC, in case anybody else wants it and doesn't follow their twitter account
[image]

ctrlaltca, 2017-05-01:
Whoops, looks like you missed a "c" in ArcticFox

ETPNV13, 2017-05-01:
Shit...I made it in the middle of the night...easy enough fix, give me a minute...

ETPNV13, 2017-05-01:
Or maybe this...kinda going for a dreamscape...
[image]

ETPNV13, 2017-05-01:
Or in greyscale maybe...
[image]

ETPNV13, 2017-05-01:
And this just for giggles...
[image]

smok-e, 2017-05-01:
very interesting pictures,
tho I would like something a bit less sinister ;-)
But it makes me curious how you did them.
how did you compose them?
did you digitally paint them or by hand?
Or did you take some pictures and composed them?
which software did you use? ...

ETPNV13, 2017-05-01:
I use GIMP, which is basically free photoshop with pretty much all the same features and functionality which you can get here https://www.gimp.org/downloads/, the backgrounds are originally by a polish artist and then digitally uploaded, although I do a lot of my own work as well I just loved these paintings very much and wanted to make them my own for a desktop background, the artist is Zdzisław Beksiński and he had a "thing" for "photographing dreams" where he would paint many such things as you see above

ETPNV13, 2017-05-01:
The pictures I uploaded are a little different than his originals and they have been resized to fit a 1920x1080 screen as well so the aspect ratio is very different from his paintings but if you have any other questions just let me know, happy to help

ETPNV13, 2017-05-01:
Oh and if you want some cool custom fonts you can find many for free at http://www.dafont.com/ which is where I get most of my fonts from for doing custom stuff and the font I used here is called "flesh wound"

ETPNV13, 2017-05-01:
@smok-e Maybe this is more your style?
[image]

smok-e, 2017-05-01:
looks cool :-)

I use GIMP too, since many years
and know dafont.com
just would be interesting how you compiose the pictures.
my avatar is done in GIMP ;-)
was from a foto I found somewhere on the net
but heavy edited and layered and ...  ;-)

ETPNV13, 2017-05-01:
I either create my own(I usually will use "colour paint" on linux to make a custom work of art)  or start with a digital copy of one I like and then resize to fit(select all and then resize image) my general purpose. If I want to do a certain color scheme I just "colorize" from the color menu and adjust the sliders until I am happy with the color and then just add text to my liking...pretty standard stuff, I like stuff with heavy black backgrounds because I find them easier to add more stuff to (or just darken the image to make any copy and pasted parts blend easier)Here's some Star Wars pics
[image]
[image]

 I did for my desktop I built....
[image]

ETPNV13, 2017-05-01:
Here's one from my old dual boot linux laptop from a few years back...
[image]

smok-e, 2017-05-02:
here is a desktop I did inspired by your work
[image]
I used some pictures by beksinski and edited and layered them in GIMP

ETPNV13, 2017-05-03:
I like it!!!

closed 2017-05-15 by TBXin