maelstrom2001/ArcticFox #300: Our Forum

TBXin, 2017-05-01:
### Question to our community
What do you think, do we need a forum? Communication, news, topics, your guides, etc.
Or may be current issue tracking is more than enough?

ETPNV13, 2017-05-01:
I vote yes!

Astrologies, 2017-05-01:
I also vote yes!

TBXin, 2017-05-01:
Thumbs Up for "YES", Thumbs Down for "NO" 😉

mumpen, 2017-05-01:
👍

TheSpiritof69, 2017-05-01:
What about reddit for something like that?

bumperiano, 2017-05-01:
👍

BigBadBlackWolf, 2017-05-01:
👍

smok-e, 2017-05-01:
👍

andreivoinea93, 2017-05-01:
:+1: for reddit sub

tintreach, 2017-05-01:
You could ask one of the big forums like ECF or VU to host a section for you and forgo the headaches of maintaining a dedicated forum.  Whatever decision happens I do believe this project needs to migrate from using the issue tracker for banter and return it back to its intended purpose.

antara33, 2017-05-01:
👍

Allineer, 2017-05-02:
👍

El-Grasa, 2017-05-03:
👍

joeblowma, 2017-05-03:
👍

grubia, 2017-05-03:
👍

OlafKroll, 2017-05-03:
👍

lordage, 2017-05-03:
i happen to be a mod at VU and Artic fox could ask admin about having a hosted section for certain... it would fall under the http://vapingunderground.com/forums/vaping-related-but-doesnt-fit-in-a-category.39/ section otherwise nothing wrong with containing all talks to a thread or a section if approved by the higher ups.

kaleem114, 2017-05-04:
I would be willing to give you forum space to use on wwnde.com just start a topic and I will make the relevant people admin etc

smok-e, 2017-05-04:
couldn't you just make an international (english) section in the 
http://www.ecigtalk.ru/forum ?

TBXin, 2017-05-04:
@smok-e unfortunately no. This is not our forum and they owners are not interested in this actions.

To all: we think about own forum that we can create and manage personally, to stick to our vision, etc.

TheSpiritof69, 2017-05-04:
Sure, makes sense.

I assume you have the ressources and capabilities to host and run your own forum software (since someone of you is running nfeteam.org I guess), otherwise I'd be happy to assist.

TBXin, 2017-05-04:
@TheSpiritof69 yeh, you are right. At our disposal there is a good virtual hosting, on which we can host any forum engine.

dedone, 2017-05-05:
👍

Dil3mm4, 2017-05-08:
The only thing that could scare you guys out, it's that the forum could be oversaturated of issues.
Like pretty much what already happens on Github. 
It could bring the whole thing to a customer-care service due to the large amount of people not reading already the issues opened over Github.
At least, in my opinion.

TBXin, 2017-05-16:
Hello here!
We are happy to announce that our forum is ready for our dear community 😉 
We will gradually fill it with content 😃 

https://nfeteam.org/forum