maelstrom2001/ArcticFox #305: Support for eVic VTC Primo 2.0 228W

derTim, 2017-05-03:
Hi, i just recieved the updated Primo 2.0 today and tried flashing 170331 nightly with the v6.5.109 tools.

NFE Toolbox recognizes it as "[E203] Unknown device", FH 4.05 HW 1.00 and won't flash the nightly ("Opened firmware file is not suitable...")

Anything i can do to help you supporting it? ArcticFox is great and i absolutly need it! :grin:

maelstrom2001, 2017-05-03:
https://github.com/maelstrom2001/ArcticFox/issues/256#issuecomment-292998166
Will be included to upcoming release, please be patient :)

ctrlaltca, 2017-05-03:
If you look at the [preliminary README](https://github.com/maelstrom2001/ArcticFox/tree/master/preliminary), it shows support for both eVic Primo and eVic Primo 2.0, so it should be supported as soon as the next version is releases.

derTim, 2017-05-03:
Sorry, missed the entry about the 2.0 in the readme and wasn't sure if it has the same hardware as the regular Primo :)

Looking forward to the release... After using AF on my VTC Dual and Aster i totaly forgot how bad the stock FW is :)

closed 2017-05-03 by derTim