maelstrom2001/ArcticFox #306: Auto Off Feature?

Extermicide, 2017-05-04:
New to ArcticFox, and LOVING IT!

 Was looking thru the settings, and was wondering if (maybe i'm missing it), there's a way to have our device/mod auto-turnoff after a set time (lets say 3 mins) - to both conserve battery and as a safety precaution if unit is placed it in pocket or vape-bag. Kinda makes sense to me that if the unit is not gonna be used - to turn it off, and unfortunately being the human creature's we are, sometimes we forget.

Cheers, and Thank-U for all the hard work.

adzam016, 2017-05-04:
+1 for this

hawkeyexp, 2017-05-04:
Would be great  +1

maelstrom2001, 2017-05-04:
Well, how about option in addition to standard deep sleep?
Now our boxes goes in deep sleep after 3 minutes of inactivity. We can easily add option to lock / power off device before it goes to sleep. In this way, if you're choosing some option, for example, "Lock on Sleep", box will be not wake up on single keypress from deep sleep, you'll have to power on it by 5 clicks. Is it suitable?

hawkeyexp, 2017-05-04:
Sounds like a good option

Tomozaki123, 2017-05-05:
me to +1

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001