maelstrom2001/ArcticFox #307: Do you love Arctic Fox?

adzam016, 2017-05-04:
Hi  Arctic Fox users!

First of all, I do not work for, endorse or even know the developers.

I just wanted to say... DONATE THEM SOME MONIES! ££££

These guys have significantly improved our vape lives, and I am sure almost all of you will agree.

We keep requesting features and finding problems but I bet 80% or more of us will never consider donating to them!

I get paid on the 25th of this month and I will be giving a healthy donation to them.

If you like ArcticFox, give these guys a bit of credit for their time. Go on their official website and chuck them some money using the link.

By donating, we will be funding (and encouraging) these guys to continue to develop the firmware as well as add new devices etc.

Show the Arctic Fox team some love

Niko1983, 2017-05-04:
I love it! I've no job, but I'm in for giving them a beer from me! They bring new life for my Wismec RX200S and I've bought the Joyetech VTCDual cause it's a supported one. If my contribution can help them, why not? :D

mumpen, 2017-05-04:
I love ArcticFox and I'll donate $10000 if Predator support comes out today :P

TBXin, 2017-05-04:
@mumpen I do not believe you 👿

Niko1983, 2017-05-04:
I'll prefer to spend a couple of bucks on VTC Dual :D

TimB87, 2017-05-05:
Can't donate much since I'm a broke ass student but I just did my part. 
Thanks for the great work guys! Eleaf/Joyetech/Wismec should donate generously since ArcticFox was my buying argument for the RX2/3 and Predator recently and I think this will apply to a lot of people!

nifelvind, 2017-05-08:
Love u guys!

closed 2017-05-15 by TBXin