maelstrom2001/ArcticFox #308: Is the eVic Primo 2.0 supported?

Justin-L-Franks, 2017-05-05:
I see in the [changelog](https://github.com/maelstrom2001/ArcticFox/wiki/Changelog) that support for the eVic Primo has been added. Does this include the eVic Primo 2.0 or only the original?

nifelvind, 2017-05-05:
The upcoming release will support eVic Primo 2.0. The actual version does not support the 2.0 yet. So be patient ;)

Justin-L-Franks, 2017-05-05:
Awesome. I'm in the market for a new dual-18650 mod that fits 24 mm and 25 mm atomizers without overhang, in order to replace my heavily-used and beat-up Cuboid, and I have come to rely on the features provided by the ArcticFox or myevic firmwares (especially the percentage-based power curves), and the eVic Primo 2.0 seems to be the best choice.

acevalsl, 2017-05-08:
sorry, and evic primo mini??

nifelvind, 2017-05-08:
As stated in the preliminary release notes, eVic Primo Mini will also be supported in the next release.

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001

Justin-L-Franks, 2017-05-09:
Awesome! Thanks so much. My new eVic Primo 2.0 will be delivered tomorrow or Thursday, and the first thing I am going to do is install ArcticFox.