maelstrom2001/ArcticFox #310: [ENH] Lower amp limits for security on off-spec batteries

a-braun, 2017-05-05:
Hi,
I wonder if it could be possible to have an option in future releases to lower the amp limit of any given hardware to fit the specs of batteries. for example, if I use a HG2 in a istick pico, so I could safely lower the maximum ampere to 20, as is the spec of the battery.
I dont want to ask for a higher amp limit, just to lower it!
The only problem i can think of is to track the max. amps per device(to not go over the actual limit), but this should be already in your firmware, as its one file for all devices...
Is this possible or too much work?
And thank you for your firmware, its way better than any stock firmware. this saved me ~ 100€ for a DNA device :)

maelstrom2001, 2017-05-05:
#32

a-braun, 2017-05-05:
So to recap, this is hardcoded in the device and we can't set these values via firmware?
Did you mean that in the comment? 
Не совсем понял ваш комментарий в #32.
Может есть время описать проблему на русском?