maelstrom2001/ArcticFox #314: Some development requests

arcticfoxer, 2017-05-06:
First of all I congratulate you. The work you do is not something everyone can accomplish. Thank you very much for converting our ordinary devices into DNA devices.

1-When we press the + key to increase the wattage, it should increase by 1 watts at an increasing speed. + Key When we press individually, 0.1 watt can be increased. It also allows you to remove the 1 watt on / off option from these settings.

2-The keys can be locked when we press the ignition key less than 5 times. Because when we press 5 times, the device is already off. In this case, there is no advantage of locking. Can you provide this?

Thank you for reading and taking your time. In this busy time, you are also trying to make time for our demands and problems. You are the people of perfection

Dil3mm4, 2017-05-06:
Mhm. 
I'll answer on what I know. 
3- It is already available as a parameter when you change your skin elements.

arcticfoxer, 2017-05-08:
3--no my friend. I meant to be added to the info section in the menu

arcticfoxer, 2017-05-18:
@maelstrom2001 is it possible