maelstrom2001/ArcticFox #321: Back Button

xandersk, 2017-05-06:
I'd really appreciate if you guys added a back button in the menu screens so that we won't have to exit the menu every time we want to get into another menu screen or if we accidentally entered the wrong menu screen and wanted to go back. It'd also help allot when editing multiple settings 😀 

PS: I sense that the long awaited new release is imminent, can't wait to get it on my predator!!!

maelstrom2001, 2017-05-06:
Hello! You can use "back" shortcut in menu navigation: hold + and - buttons (or Fire and - button) simultaneously.

xandersk, 2017-05-06:
@maelstrom2001 Sorry for the duplicate, forgot to delete the is:open form the search form when i searched for a similar issue, silly me 😞 

Good job on the firmware, i wish joyetech would hire you or at least give you access to unreleased firmware s. c. Having Arctic as stock firmware would definitely increase the popularity of their mods 🏆

closed 2017-05-08 by maelstrom2001