maelstrom2001/ArcticFox #323: Evic Primo 1.0 atomizer issue

kfks, 2017-05-07:
Hello, iam getting atomizer short on every tank i tried after i installed af_170331. It works with no issue when i upgrade it back to stock 4.05 fw for evic primo 1.0. Does anybody have the same issue with AF for evic primo 1.0?

kfks, 2017-05-07:
update: the issue with atomizer short on af_170331 is only when i use artic configuration

maelstrom2001, 2017-05-07:
Hello!
It's a known issue on Primo, and it is 100% hardware related. Negative contact goes through body, 510 thread is press fit and may loose good contact to ground. The best solution is to solder wire to negative post of connector and connect it to common ground.

[image]
[image]

kfks, 2017-05-07:
I know this is mostly HW issue but my Primo works with no problems when using 4.05 original fw. Issue is only with AF installed.

maelstrom2001, 2017-05-07:
It is a common problem. Yes, the stock firmware produces an "atomizer short" error less often than AF,  'cause it have weak overcurrent protection. On other side, I can not reproduce this error on my early Primo, so I can not guarantee that this hardware issue can be fixed by software.

roaminghawk, 2017-05-07:
So the primo's 510 is also press fitted like the predator? 
What about the primo V2? 
Also do you have any info on the Primo Mini?

xandersk, 2017-05-07:
It's not exactly like the predator (the diameter of the press fitted threads is bigger on the predator) but yes, it's press fitted. Do you get Atomizer Short or No Atomizer Found? 

- For No Atomizer Found search for Electrically Conductive Epoxy/Adhesive and use it to glue the 510 to the casing. Make sure that both surfaces are thoroughly cleaned. Sand any paint that might be present on the interior of the casing near the area of the 510 hole and clean again. Do not use too much glue, just enough to fill the space between the threads and the casing.

- For Atomizer Short, 
     - Try using another atomizer. Same issue? 
     - Did you check the resistance of the atomizer with an ohm meter?
     - Try to extend the 510 positive pin of the atomizer. (if it's adjustable)

Arctic Fox has a more sensitive approach to reading the resistances than stock, if your atomizer is bordering the 0.10 OHM mark then Arctic might reject it saying that it shorted. My Stock Firmware Predator228 also reads the resistances a bit higher than my AF VTC Mini.

kfks, 2017-05-07:
Thank you for info :)

maelstrom2001, 2017-05-07:
> So the primo's 510 is also press fitted like the predator?

Press fitted too, but connectors are different. Predator's connector consists of two halves, and the top part is problematic. 510 on Primo AFAIK is monolith, but I think it is press fitted without removing the paint from opening for it, so there is no good contact between body and 510. Also problem can be in the mounting screws of the PCB, which supply common ground to the housing. The most problematic Primos are black painted as I saw on Russian forum.

roaminghawk, 2017-05-08:
Thanks for the response. 
In your opinion, what's the best made (Quality of parts, fast processor, etc) mod that supports AF?

andreivoinea93, 2017-05-08:
I'm also facing this issue.

So the Primo does not have a ground connected directly to the 510 connector ground? Does it have only the positive and the negative goes from somewhere else in the casing? Damn bad design, Joyetech.

Please confirm if I understood that correctly, if so I'll probably go and fix it.

hunterj12, 2017-05-08:
Does the primo 2.0 have the same 510 issue as the original primo?

andreivoinea93, 2017-05-09:
Ok so I tared down my Primo and indeed the problem is what I expected. Can you please tell me what should I solder the ground to? Also, does it need to be a big wire?

andreivoinea93, 2017-05-09:
as an update, I tried to fix this but solder won't stick to the 510 ground. Any other ideas?

komaramnt, 2017-05-10:
Use sandpaper to sand a little spot that you want to solder and use flux.
If you are not soo good whit solderig view basic soldering videos on the web.

kfks, 2017-05-14:
Hello, so the news
My problem is (propably) solved with new AF version 170510, my mod works fine now with no issues. I get atomizer short once but i solved it by turn mod off and on. :)